usnistgov/ARIAC #1: controllers not available

dan9thsense, 2020-02-05:
Running the example with
`roslaunch nist_gear sample_environment.launch`

The environment loads ans shows up in Gazebo OK ans lots of rostopics are being published, but it cannot find the gantry and arm controllers.

Here are the errors:
```
[ERROR]: Could not load controller 'gantry_controller' because controller type 'effort_controllers/JointTrajectoryController' does not exist.
[ERROR]: Use 'rosservice call controller_manager/list_controller_types' to get the available types

[ERROR]: Could not load controller 'left_arm_controller' because controller type 'effort_controllers/JointTrajectoryController' does not exist.

[ERROR]: Could not load controller 'right_arm_controller' because controller type 'effort_controllers/JointTrajectoryController' does not exist.
```

Here is the result of the suggested call, 
`rosservice call controller_manager/list_controller_types`

`ERROR: Service [/controller_manager/list_controller_types] is not available.`

Pavel-P, 2020-02-05:
You need to install `ros-melodic-ros-controllers` to get those controllers types. I've updated the package manifest to reflect this.
The dependencies haven't changed from last year's ARIAC, so you can find the complete list by running `apt depends ariac3`

dan9thsense, 2020-02-05:
That fixed it.

closed 2020-02-05 by dan9thsense