usnistgov/ARIAC #2: Logical Camera Definition Broken

leegs, 2020-02-06:
Hello,

I had a significant issue getting the sample_environment to run.  What finally fixed it was altering the "collision" and "visual" elements of the logical camera definition in ariac.world.template.  The original text was looking for a mesh from a uri this way: "model://logical_camera/meshes/camera.dae."  The logical camera model in nist_gear is named logical_camera_ariac.  I altered the uri to: "model://logical_camera_ariac/meshes/camera.dae" and the sample_environment works now.

I hope this is helpful.
[thediff.txt](https://github.com/usnistgov/ARIAC/files/4167848/thediff.txt)

zeidk, 2020-02-07:
Hello,
this is strange that you had to modify any file to make this work. Can you please post the part you modified here?

Thank you,

zeid

Pavel-P, 2020-02-07:
Thanks for the catch, it turns out this broken definition was part of a deprecated quality_control_sensor. Should be removed now.

leegs, 2020-02-07:
I'm glad I could help.

closed 2020-02-07 by leegs