usnistgov/ARIAC #3: ariac-docker repository

dan9thsense, 2020-02-17:
The [tutorial for preparing submissions](https://github.com/osrf/ariac-docker/blob/master/README.md) links to an osrf repository that was used last year.  Is that still valid for this year?  If not, is there an updated repository for ariac-docker?

Pavel-P, 2020-02-24:
No, we still need to update that repository to use this year's docker images, but once that's done we'll update the documentation and let you know.

zeidk, 2020-02-24:
We will update the documentation on docker by the end of this week. We will also try to document the other parts of the wiki as well.

dan9thsense, 2020-03-02:
Did this get done?  The docs still point to the osrf repo.  I'm concerned that there is only a month left until the qualifiers and it often takes some time to get docker stuff working correctly.

zeidk, 2020-03-02:
Hi Dan,
we are in discussion with OSRF about the docker image to use. I will update the webpage and reply here as soon as I hear from them.

thank you,

zeid

zeidk, 2020-03-09:
Just an update that the docker image will be pushed tomorrow. The tutorial on automated evaluation will be updated accordingly.

dan9thsense, 2020-03-13:
I couldn't find the new image; looks like the tutorial still points to the old pages.

zeidk, 2020-03-13:
We are working on the dockerhub account. I will update this thread when this gets fixed.

zeidk, 2020-03-21:
Docker images were pushed on dockerhub and bash scripts were updated.

Please see wiki/documentation/automated_evaluation.md and wiki/tutorials/automated_evaluation.md

Closing this ticket for now. Open a new one if any issue arises with automated evaluation.

closed 2020-03-21 by zeidk