usnistgov/ARIAC #4: Collision issues causing massive slowdown

stevegray954, 2020-02-23:
Lifting a single part is giving me a speed drop from 0.98x to 0.11x.  Until the part has been resting long enough that Gazebo 'sleeps' all the collision checking on it, the sim speed remains very low.

[Video on Youtube](https://youtu.be/LZqKVDVQfMs)

stevegray954, 2020-02-23:
I had moved the gear to another location, but I can confirm this happens if you just lift it above the bin it starts in.

stevegray954, 2020-02-23:
On further investigation, issue is caused by `libROSAriacTaskManagerPlugin.so`
(I removed lines 344-416 from `nist_gear/worlds/ariac.world.template` and the slowdown is gone)

stevegray954, 2020-02-24:
Worth noting that neither before nor after the fix was the system recognizing contact between the arms, because they are allowed to pass through eachother.

stevegray954, 2020-02-24:
Workaround is present as PR #7

Pavel-P, 2020-02-24:
Thanks for taking a thorough look at this. As a side note, we hadn't updated the collision filter to use the "left_" and "right_" prefixes for the link names, which was contributing to the slowdown a lot. But even after fixing that your approach stills seems to be noticeably faster, so I'll go forward with the PR.

closed 2020-02-24 by Pavel-P