usnistgov/ARIAC #5: Product shelf locations

stevegray954, 2020-02-23:
Looking at the [competition spec](https://github.com/usnistgov/ARIAC/blob/master/wiki/competition_specifications.md), it looks like the shelves can be in a large number of configurations.  Will these be known before we specify the locations of the sensors in the world?

Pavel-P, 2020-02-24:
Currently, gear.py is designed to take the environment and sensor configs simultaneously so I don't think that we could do this without making significant modifications to both it and our submission system.
We're currently looking into a better way of allowing statically-placed sensors to reliably detect all needed parts across all shelf configurations. You can expect an update on this sometime next week.

stevegray954, 2020-02-26:
Looking forward to it.  As it stands, if we use 2 logical cameras to cover each possible shelf location, we're guaranteed to have a camera smack in the middle of each gap.

stevegray954, 2020-03-02:
Practically speaking, do you need to change the locations of the shelves?  Since we have to specify static camera locations, it seems weird that the shelves might be in different positions.  If we could place sensors on the robot, it would be a different matter.

Pavel-P, 2020-03-03:
The changing shelf locations will make a bit more sense with our next update. We'll be introducing an obstacle avoidance aspect to the challenge that will involve using the gaps between shelves to navigate the environment. The main reason for the changing locations is to encourage teams to incorporate some form of dynamic path planning into their design.

stevegray954, 2020-03-06:
Any idea when the next update will be?  It is getting late to add changing shelf locations and simulated people walking around.

zeidk, 2020-03-06:
We are in the process of moving our source from the test repository to this one. This will be done today. The documentation on competition specifications will have more details on shelves.

zeid

Pavel-P, 2020-03-06:
The update I mentioned should be available now. Just to recap changes that are pertinent to this issue:
* The farthest group of 3 shelves are now static across all runs
* No parts will spawn on shelves that move
* Shelf locations will NOT be provided before run time
* ALL shelves and bins now publish their transforms during run time

If there's still any confusion about how this works let me know.

stevegray954, 2020-03-06:
That is clear. Thanks!

closed 2020-03-27 by stevegray954