usnistgov/ARIAC #6: All contacts go active once anything moves

stevegray954, 2020-02-23:
Check out this video of moving a gear:
https://www.youtube.com/watch?v=d4r4zSL8EOg&feature=youtu.be

All the other contacts on all other objects activate.  This shouldn't happen.  Unfortunately, it's likely a gazebo bug.

Easy to replicate - turn on contact visualization and move anything.

stevegray954, 2020-02-24:
I can replicate this issue with just plain Gazebo 9.12 by adding two boxes and looking at contacts when I move one box.  All contacts go active; this is a Gazebo issue and not an ARIAC specific issue.

Pavel-P, 2020-02-24:
I've ran into this issue in the past-- however I think it only happens when manually changing the locations of parts in the GUI. Interacting with parts using the robot or conveyor should never result in contacts re-activating for all parts, in my experience. Let me know if you run into this issue while running the competition (with `moveit_example.py` for example) and I'll take another look at it.

closed 2020-02-24 by Pavel-P