usnistgov/ARIAC #8: No self-collision

stevegray954, 2020-02-24:
The arms are allowed to pass freely through each other as well as through the base.  This means that the check for arm-arm collision in `RosAriacTaskManagerPlugin::OnContactsReceived()` doesn't register when the arms collide.

To replicate: use `rqt robot_description:=/ariac/gantry/robot_description` to pass the arms through eachother.  The line `ROS_ERROR_STREAM("arm/arm contact detected: " << contact.collision1() << " and " << contact.collision2());` in OnContactReceived() should print to ROS ERROR

stevegray954, 2020-02-24:
Will create a fix to allow self collision checking by updating the urdfs/xacros.

Also noticed that now all robot components are prefaced by the `gantry`
```[Dbg] [ROSAriacTaskManagerPlugin.cc:1218] gantry::left_forearm_link::left_forearm_link_collision with gantry::torso_main::torso_main_collision```

This means we can check for the word `gantry` in the collision strings to check for robot self-collision in `OnContactReceived`.  This should speed things up a bit.

stevegray954, 2020-02-24:
Yup, got it to fire with urdf fixes.  Now checks for arm-arm and arm-torso collisions.

```[Dbg] [ROSAriacTaskManagerPlugin.cc:1218] gantry::left_forearm_link::left_forearm_link_collision with gantry::right_wrist_1_link::right_wrist_1_link_collision
[ERROR] [1582507283.761705552, 195.195000000]: arm/arm contact detected: gantry::left_forearm_link::left_forearm_link_collision and gantry::right_wrist_1_link::right_wrist_1_link_collision
```

Pavel-P, 2020-02-24:
Thanks, looks good. I'll also update the gantry dimensions to account for collisions between the torso_base and rails in a moment.

closed 2020-02-24 by Pavel-P