usnistgov/ARIAC #10: Qualifier info and scenarios

stevegray954, 2020-02-29:
Are there any more details you can share about the qualifier?  April is just a month away we are short on details.  Even dates in April would be useful for time budgeting.  When can we expect finalizing competition details like sensor configurations and shelf layouts?  What about releasing qualifier scenarios? When will the deadline be?

zeidk, 2020-03-02:
Hello Steve,

The important dates can be found at https://www.challenge.gov/challenge/ariac/ but I posted them on the wiki home page.

We are doing this for the first time ourselves and we would like a little bit more information on what OSRF provided the previous years.

- In which format did you use to receive details about sensor configurations and shelf layouts? Were they just writeups or  ARIAC config files?
- We are working on qualifier scenarios but will be able to release some trial config files for qual1 this week.

I will keep you updated,

thank you

stevegray954, 2020-03-02:
Thanks for the dates.
- Competitors provided details about their sensor configurations in a [competitor configuration file](https://github.com/usnistgov/ARIAC/blob/master/wiki/configuration_spec.md)
- No information on shelf or bin layouts was provided aside from the wiki description.  The shelves and bins never moved, and we could look at the Gazebo world file to find out exact locations.
- No information on the contents of bins was provided.  We had to get that from our sensors.  (There was an  `osrf_gear/GetMaterialLocations` service for development, but it was not allowed in competition)

zeidk, 2020-03-02:
Hi Steve,
I will bring up these dynamic shelf placements during our group meeting tomorrow. We will also discuss about mounting a sensor on the robot. Any type of sensor you have in mind (beside a logical camera)?

zeid

stevegray954, 2020-03-03:
How about a RGBD camera?

dan9thsense, 2020-03-06:
It's WAY too late to add a sensor to the robot!  It's also too late to add an RGBD camera.  Those were both good ideas before the competition started, but at this point, a lot of development hours would be wasted if either of those changes were introduced.

Pavel-P, 2020-03-06:
The update we pushed today should be the last major change regarding sensors and what to expect from qualifier scenarios. This means that we won't be adding any sensors to the robot this year and the sensor configuration format will remain unchanged from last year's. As mentioned in #5 we won't be providing shelf locations beforehand, but they can be determined at run time through TF.

We also released some additional sample configurations (`sample1-4.yaml`) that are representative of the qualifier scenarios. At the top of these configs there are comments that mention the high-level challenges that the config is designed to test. The qualifiers will test the same high-level challenges but will differ as far as part types, locations, quantities, etc. I'm not sure if this is properly documented in the wiki yet. If not, we'll fix that ASAP.

As far as your point on the RGBD camera @dan9thsense do you mean on the robot or in general? I've updated the existing ARIAC depth camera to behave like a true RGBD camera but I can revert this if there's a good reason to do so. As it stands I don't see any way that switching the depth camera to RGBD would invalidate/undermine any existing approaches to the challenge. We should move this part of the discussion to a new issue though.

If there's still any ambiguity regarding the qualifiers, let us know.

stevegray954, 2020-03-06:
Are you still planning on adding people walking around?  If so, when would that be?

Pavel-P, 2020-03-06:
That is included as part of this update as well, though I'm not sure how clearly things are documented in the wiki currently. As a quick summary, up to 2 moving people are included in some of the more difficult scenarios and will move back and forth along the rows of shelves in a periodic motion, stopping at either end of the row. The timing for when each person moves and stops is NOT revealed beforehand. `sample_obstacles.yaml` gives an example of how people might behave in the environment.

zeidk, 2020-03-06:
Documentation on moving obstacles and shelves are now in the configuration specifications page (https://github.com/usnistgov/ARIAC/blob/master/wiki/competition_specifications.md)

dan9thsense, 2020-03-07:
@Pavel-P Changing the depth cameras to RGBD cameras makes a huge difference in the competition.  I was originally hoping to use RGBD cameras this year.  Their absence pushed me to use the digital cameras and A LOT of work has gone into getting things running with the digital cameras.  To go back and redo that would be very painful, yet necessary, due to the cost difference and the large numbers of cameras needed.  I strongly feel that since the sensors were specified at the start of the competition, they should not be changed mid-way.  That imposes a big penalty to teams that focused on camera code vs teams that focused on other aspects of the environment.

dan9thsense, 2020-03-07:
@Pavel-P  I just noticed your comment that the RBGD topic should be new issue, so I opened one and copied the above comment there.

closed 2020-03-27 by stevegray954