usnistgov/ARIAC #11: Github wiki edits generate a ton of commits

stevegray954, 2020-03-06:
I'm guessing you are editing the wiki directly on Github.  Please don't.  It makes the commit history difficult to skim through to determine what has changed as we get things like 10 successive commits labeled 'Update automated_evaluation.md'.  It's easier when the commit description is meaningful (and there aren't a ton of them).

Thanks!

Pavel-P, 2020-03-06:
Sorry about that, the git practices have been all over the place for this project. I'll try to go through and squash the majority of those wiki commits when I get the chance.

zeidk, 2020-03-09:
Sorry about this, we will make sure to use better ways to edit the wiki.
Closing this ticket for now.

closed 2020-03-09 by zeidk