usnistgov/ARIAC #12: sensor change mid-competition

dan9thsense, 2020-03-07:
@Pavel-P Changing the depth cameras to RGBD cameras makes a huge difference in the competition. I was originally hoping to use RGBD cameras this year. Their absence pushed me to use the digital cameras and A LOT of work has gone into getting things running with the digital cameras. To go back and redo that would be very painful, yet necessary, due to the cost difference and the large numbers of cameras needed. I strongly feel that since the sensors were specified at the start of the competition, they should not be changed mid-way. That imposes a big penalty to teams that focused on camera code vs teams that focused on other aspects of the environment.

If you offer the RGBD cameras as a new sensor, with the same cost as the digital cameras, that would be OK. It would allow teams that want to use RGBD cameras to do so, without forcing teams that have already invested tons of time into code for the digital cameras to switch to RGBD.

Pavel-P, 2020-03-09:
If I understand correctly, you're suggesting that using RGBD cameras would be the dominant approach to the competition due to their lower cost. But doesn't the same logic apply to the existing depth camera? I'm curious why you'd feel forced to switch to the RGBD camera but not the depth camera.

The main thing we're interested in is introducing at least one cost-saving alternative to the logical camera for color identification. Are you categorically opposed to this? If not, what makes the RGBD more desirable than the depth camera?

dan9thsense, 2020-03-09:
Because the depth camera does not provide color information which is needed to pick the correct part.  You can use the depth camera to find the right type of part and carry it to a location with a logical camera (perhaps mounted in some central area) to check the color.

Pavel-P, 2020-03-12:
After giving this some more thought, I have to concede that at this point virtually any sensor change (including this one) has the potential to put an unfair burden on some teams. It's not ideal, but for now I'll go with your suggestion of introducing the RGBD camera separately at the same cost as the logical camera.

dan9thsense, 2020-03-13:
Appreciate your consideration on this.

closed 2020-03-13 by dan9thsense

RyodoTanaka, 2020-03-17:
@Pavel-P 
Hi, I have a question on `depth_camera` and `rbbd_camera`.
I understand that you update `depth_camera` can only publish PointCloud topics.
Before the update, we could subscribe also RGB data.
But because of RGBD camera update, there are no sensor can publish RGB data except `logical_camera` and `rgbd_camera`.
I can apply this modify and I understand your great effort for making the rule fair.
But I am wondering that this year's task, we need to detect the color.
And we have no choice to detect them with no sensor except most expensive cost sensors(rbgd_camera and logical_camera).
Thus, if possible, we would like to use simple rgb camera as low cost sensor ($100 cost sensor).
I think, in the following source code, simple rgb camera can use. But I could not launch them with `type: camera` parameter.
https://github.com/usnistgov/ARIAC/blob/master/nist_gear/script/gear.py#L62
In anyway, I would like to ask you about the answer for using simple rgb camera.

Thank you for your kind.
Best Regards,
Ryodo