usnistgov/ARIAC #13: right should pan joint limited range

dan9thsense, 2020-03-08:
I've encountered a weird bug and am wondering if anyone else is seeing this.  For some reason, the right shoulder pan joint only has a range of 0 - 2pi instead of -2pi to 2pi.  To see this, just use rqt immediately after startup.  rqt shows a range of -2pi to 2pi in the slider bar, but moving past 0 to negative values does not get a response from the arm; it just stops at 0.  The left arm works fine, as do all the other joints on the right arm.

stevegray954, 2020-03-08:
I can't reproduce that issue.  Started the sim and rqt up without anything else.  Right pan tracked as I moved it to negative values.

dan9thsense, 2020-03-08:
It's so strange.  I am unable to upload a video here, but these images show the problem.  In one, the shoulder is panned to .42, the other shows rqt trying to pan it to a negative value, but it is stuck at 0.

[image]
[image]

dan9thsense, 2020-03-09:
Tried re-cloning from the git repo, but the issue persists.

dan9thsense, 2020-03-31:
This problem is still present.....

Pavel-P, 2020-04-01:
I also couldn't reproduce this. Also reviewed the urdf and controller configs throughout the project and couldn't find anything resembling a joint limit of 0. The only thing that comes to mind is you might have some extraneous ur10 urdf macro with different joint limits somewhere in your workspace.

dan9thsense, 2020-04-02:
I can't find any other ur10 sources or macros on the local machine, but I understand that there is no more you can do if it is not reproducible.  When I get docker replay working, I'll test it there.  Thanks for looking into it.

closed 2020-04-02 by dan9thsense