usnistgov/ARIAC #14: Warnings -- 'Model already exists'

stevegray954, 2020-03-09:
Turned on verbose mode and see this with `sample.yaml` and `sample_obstacles.yaml`.  I haven't tried other configs yet.

```bash
[Wrn] [World.cc:1079] Model with name [bin4|piston_rod_part_blue_6] already exists. Not inserting model. This warning can be ignored in certain situations such as rewind during log playback.
[Wrn] [World.cc:1079] Model with name [bin4|piston_rod_part_blue_5] already exists. Not inserting model. This warning can be ignored in certain situations such as rewind during log playback.
[Wrn] [World.cc:1079] Model with name [bin4|piston_rod_part_blue_4] already exists. Not inserting model. This warning can be ignored in certain situations such as rewind during log playback.
[Wrn] [World.cc:1079] Model with name [bin4|piston_rod_part_blue_3] already exists. Not inserting model. This warning can be ignored in certain situations such as rewind during log playback.
[Wrn] [World.cc:1079] Model with name [bin4|piston_rod_part_blue_2] already exists. Not inserting model. This warning can be ignored in certain situations such as rewind during log playback.
[Wrn] [World.cc:1079] Model with name [bin4|piston_rod_part_blue_1] already exists. Not inserting model. This warning can be ignored in certain situations such as rewind during log playback.
[Wrn] [World.cc:1079] Model with name [bin3|gear_part_green_2] already exists. Not inserting model. This warning can be ignored in certain situations such as rewind during log playback.
[Wrn] [World.cc:1079] Model with name [shelf1|gear_part_red_5] already exists. Not inserting model. This warning can be ignored in certain situations such as rewind during log playback.
[Wrn] [World.cc:1079] Model with name [shelf1|gear_part_red_4] already exists. Not inserting model. This warning can be ignored in certain situations such as rewind during log playback.
[Wrn] [World.cc:1079] Model with name [bin3|gear_part_green_4] already exists. Not inserting model. This warning can be ignored in certain situations such as rewind during log playback.
[Wrn] [World.cc:1079] Model with name [bin3|gear_part_green_1] already exists. Not inserting model. This warning can be ignored in certain situations such as rewind during log playback.
[Wrn] [World.cc:1079] Model with name [shelf1|gear_part_red_1] already exists. Not inserting model. This warning can be ignored in certain situations such as rewind during log playback.
[Wrn] [World.cc:1079] Model with name [bin3|gear_part_green_3] already exists. Not inserting model. This warning can be ignored in certain situations such as rewind during log playback.
[Wrn] [World.cc:1079] Model with name [shelf1|gear_part_red_3] already exists. Not inserting model. This warning can be ignored in certain situations such as rewind during log playback.
[Wrn] [World.cc:1079] Model with name [shelf1|gear_part_red_2] already exists. Not inserting model. This warning can be ignored in certain situations such as rewind during log playback.
```

Is this expected behavior?

dan9thsense, 2020-03-09:
I'm seeing some cases where the model remains visible on the bin after being picked up by the gripper.   Wonder if that is related?

zeidk, 2020-03-09:
Thank you for the feedback. Pavel or myself will look into this tomorrow.

Pavel-P, 2020-03-09:
Thanks, should be fixed in the latest commit.

closed 2020-03-09 by Pavel-P