usnistgov/ARIAC #15: Gantry only spawns if 'development_mode' is set to true

stevegray954, 2020-03-09:
I noticed in the sample launch file that development mode is set to true.  Removing the `--development-mode` flag cause the gantry not to spawn.

I had thought all the `development-mode` flag did was enable the cheat topics on the [interface page](https://github.com/usnistgov/ARIAC/blob/b9afa1389f5232c48b85469b0a1397436471d0b7/wiki/competition_interface_documentation.md), but it seems to be doing something else.

```
  <node name="ariac_sim" pkg="nist_gear" type="gear.py"
        args="--development-mode
          $(arg verbose_args)
          $(arg state_logging_args)
          $(arg gui_args)
          $(arg load_moveit_args)
          $(arg fill_demo_shipment_args)
          --visualize-sensor-views
          -f $(find nist_gear)/config/sample.yaml
          $(find nist_gear)/config/sample_user_config.yaml
          " required="true" output="screen" />
```

Pavel-P, 2020-03-09:
Make sure you're using the custom version of gazebo_ros_pkgs mentioned in the [wiki](https://github.com/usnistgov/ARIAC/blob/master/wiki/tutorials/installation.md), the debian package blocks the spawn_urdf_model service from completing when ros networking is disabled in competition mode.

stevegray954, 2020-03-11:
Thanks, adding the nonstandard `gazebo-ros` version fixed it.

For anyone looking at it later, the solution was to clone this into your catkin workspace:
```
git clone https://github.com/osrf/ariac-gazebo_ros_pkgs -b ariac-network-melodic
```

closed 2020-03-11 by stevegray954