usnistgov/ARIAC #17: error when submitting agv

dan9thsense, 2020-03-10:
When I call the service to submit a shipment, error messages occur.  The shipment gets submitted successfully, but I am concerned that something I am doing wrong is triggering these messages.

[ERROR] [1583860898.686263548, 31.877000000]: agv1::kit_tray_1::kit_tray_1::tray: Handle get content service called by: /gazebo
[ERROR] [1583860898.695310152, 31.885000000]: [INFO] AGV 'agv1' delivery triggered
[ INFO] [1583860899.461913803, 32.637000000]: AGV successfully triggered.

zeidk, 2020-03-10:
The service has been executed properly. The error is because of the wrong use of ROS_ERROR_STREAM (see here https://github.com/usnistgov/ARIAC/blob/72d2134545e3eb82e25d631b98daf1543a3ff02e/nist_gear/src/ROSAGVPlugin.cc#L356)
We will fix this.

We will also take a look at the first error message you got.

zeid

Pavel-P, 2020-03-12:
Thanks, fixed in adfcb843b1426c1407014404d5eb81b41e9fa54c.

closed 2020-03-12 by Pavel-P