usnistgov/ARIAC #19: updated order not sent until original order submitted

dan9thsense, 2020-03-16:
Working on sample2.yaml, the updated order does not arrive until immediately after the original order is submitted.  The original order had 5 parts and the update conditions are:

announcement_condition: wanted_products
    announcement_condition_value: 4

Changing  announcement_condition_value to 2 makes no difference, the order update still does not happen until after submission of the filled first order.

Changing the condition to time works OK.

dan9thsense, 2020-03-17:
The problem is that the original order does not fill enough wanted parts to trigger the updated order.  Modifying the wanted products fixes the problem

closed 2020-03-17 by dan9thsense

zeidk, 2020-03-17:
Thank you Dan.