usnistgov/ARIAC #20: Question on RGBD camera

RyodoTanaka, 2020-03-17:
I could not re-open the following issue.
https://github.com/usnistgov/ARIAC/issues/12#issuecomment-599928372
Thus, I will copy and paste my question on here.

@Pavel-P
Hi, I have a question on depth_camera and rgbd_camera.
I understand that you update depth_camera can only publish PointCloud topics.
Before the update, we could subscribe also RGB data.
But because of RGBD camera update, there are no sensor can publish RGB data except logical_camera and rgbd_camera.
I can apply this modify and I understand your great effort for making the rule fair.
But I am wondering that this year's task, we need to detect the color.
And we have no choice to detect them with no sensor except most expensive cost sensors(rbgd_camera and logical_camera).
Thus, if possible, we would like to use simple rgb camera as low cost sensor ($100 cost sensor).
I think, in the following source code, simple rgb camera can use. But I could not launch them with type: camera parameter.
https://github.com/usnistgov/ARIAC/blob/master/nist_gear/script/gear.py#L62
In anyway, I would like to ask you about the answer for using simple rgb camera.

Thank you for your kind.
Best Regards,
Ryodo

RyodoTanaka, 2020-03-23:
Could someone please answer this ... ?

zeidk, 2020-03-31:
We apologize for not replying to this issue yet. We will try to provide an answer today.

Pavel-P, 2020-04-01:
The previous version of the depth camera was implemented as a laser scanner, so I don't think it was possible to subscribe to any RGB data prior to introducing the RGBD camera. I investigated where that `type: camera` parameter came from and I couldn't find any associated sensor with it in previous competitions, so I think it is there on accident.
Besides that, I completely agree with your reasoning about needing a way to detect color with a sensor other than the logical camera (this was the original reason for adding the RGBD camera). I also agree that there is no reason to choose an RGBD camera over the logical camera at the current cost. The only reason I chose this solution was because we failed to specify ahead of time that we are introducing additional sensors this year. Because of this, we can't introduce any additional sensors (including RGB) that cost less than the logical camera without creating unfair conditions.

zeidk, 2020-04-15:
I would like to close this one to keep things going. @RyodoTanaka Did we answer your questions?

RyodoTanaka, 2020-04-20:
@zeidk @Pavel-P
Sorry for late reply.
And thank you for your kind reply.
I understand and accept the answer.
I will close this issue.

closed 2020-04-20 by RyodoTanaka