usnistgov/ARIAC #22: Qualifier information,

bankh, 2020-03-18:
Based on the earlier discussion, I am suggested to post my question here. We are asked to submit the emails of the team members for adding them to the team's competition page. With that I have a quick question:

Are we able to update the team members after submission of the contributor's list? Or is the list fix until the end of the competition?

anthonydowns, 2020-03-18:
Are you asking to update the emails associated with the sbmissions?  or to update the info on the Google Form when you first indicated interest?

bankh, 2020-03-18:
I am asking based on the email that I received. The email was asking the
email addresses of the team members.
So the original question I have, are we able to update the team members who
can access to Github repository of the team?
However, you can interpret this question as updating the team members
during the qualifiers.

anthonydowns, 2020-03-18:
ok, for what you are asking, simply reply to that same email with additions to the emails you want added and we will take care of it!

bankh, 2020-03-18:
thanks. will do.

closed 2020-04-04 by zeidk