usnistgov/ARIAC #24: Pulley part slips from gantry torso tray

avidacs, 2020-03-20:
The pulley part easily slips from the gantry torso's tray even if the robot moves with ridiculously low acceleration. It feels like the friction coefficient was zero. On the other hand, other parts remain on the gantry tray when the robot moves with reasonable accelerations. Is that behavior deliberate?

(Suggestion: increasing the friction on the gantry tray would help a lot when using the robot to collect as many parts as possible, and not just delivering parts two-by-two.)

zeidk, 2020-03-21:
Thank you, we will change the friction as soon as possible.

dan9thsense, 2020-03-21:
Are you going to increase friction for all the parts or just for pulleys?

zeidk, 2020-03-22:
It will only be for pulleys unless someone reports other parts slipping in the tray.

dan9thsense, 2020-03-22:
Oh, all the parts slip to some extent.  I was wondering if a camera would be needed that the gantry would go under to examine where the parts ended up.  If the parts are not meant to slip, then the friction needs to be increased on all of them.

zeidk, 2020-03-28:
I have made some changes to only the red pulley for testing and **I will push it tomorrow**. 

The pulley was slipping way too much and will not even stay in the tray.  I increased the friction on the red pulley. It will still slip at full speed but will stay longer before falling. Decreasing the acceleration when pulleys (or any other parts) are placed in the tray should keep your parts in the tray but they will still move in the tray (a sensor may be needed to find where your parts are in the tray).

After you spawn the robot, place a red pulley in the tray with the command below then move the robot around and see how it goes.

`rosrun gazebo_ros spawn_model -sdf -x -0.231072 -y -0.083964 -z 1.288900 -R 0 -P 0 -Y 0 -file `rospack find nist_gear`/models/pulley_part_red_ariac/model.sdf -model pulley_part_red_5`

avidacs, 2020-03-28:
Sounds promising, thanks! No problem if the parts start to slip when robot acceleration is high. That's just part of the game :) i'll try this tomorrow when you push the code.

zeidk, 2020-03-30:
Thank you,
please test it and let me know.

zeid

avidacs, 2020-03-30:
Hm... I still experience strange behavior. A few times when the pulley "stucks" then it remains more or less on the gantry tray, only slightly slips. That's fine. But in most of the cases the red pulley starts to slip even if the gantry is moved with a very low max acceleration. Immediately when the robot starts to crawl, the pulley remains in its actual location, slowly but surely slipping off the tray. I tried to attach a small screen capture on this phenomenon. 

[red_pulley_slips.zip](https://github.com/usnistgov/ARIAC/files/4404007/red_pulley_slips.zip)

zeidk, 2020-03-30:
Thanks for the feedback. We will work on modifying the friction of tray itself instead of the parts in this case.

avidacs, 2020-03-31:
Yes, I do, I double checked it. Now I see what's happening: even if I don't move the gantry the pulley is slipping on the tray very-very slowly. It only has contact with the gantry on a tiny area since the pulley has a small bump on its bottom rim. Once it slips to the edge and the small bump reaches and the rim and falls over, the pulley settles on the tray on a much larger surface. If I move the gantry only then the remains more or less in its place. But clearly there is no friction between the tray and the (small bump on) the pulley. (I remember that pulleys behaved the same in ariac 2019 as well, slipping in a slanted bin until reaching the brim of the bin.)

zeidk, 2020-03-31:
I did increase the friction on both markers (tiny areas).
One solution is to rotate the pulley before placing it on the tray so the marker is outside the tray.
The following command will place the pulley on the tray so that the marker is outside the tray. Moving the robot with the pulley placed this way will provide better results. If it's still bad then the final solution would be to not put the pulley on the tray (even increasing the friction of the tray will not really help).

``rosrun gazebo_ros spawn_model -sdf -x -0.276959 -y -0.025406 -z 1.288900 -R 0.000005 -P -0.000026 -Y 1.644586 -file `rospack find nist_gear`/models/pulley_part_red_ariac/model.sdf -model pulley_part_red_5``

avidacs, 2020-03-31:
Thanks for the suggestion, and yes, the pulley stays on the tray when its marker is outside. That's fine with me, I'll be careful how to carry pulleys on the tray. Thank you for your effort! You might consider this issue closed :)

zeidk, 2020-03-31:
Awesome! Thank you for testing :)

zeid

closed 2020-03-31 by zeidk

stevegray954, 2020-04-02:
@zeidk The green and blue pulleys no longer match the red pulley.  Are you going to undo the red changes or update the green and blue?

stevegray954, 2020-04-03:
@zeidk Yes, please undo all the changes on the red pulley. With those changes, the pulley vibrates around (away from where it was placed - making it impossible to get the points for correct pose).

Here's what I see when placing it in `qual_a_3` (after I added red pulleys in with my PR #39):

```
rosrun tf2_tools echo.py kit_tray_1 lc_tray1_pulley_part_red_4_frame
At time 30.05, (current time 30.636)
- Translation: [0.103, -0.191, 0.079]
- Rotation: in Quaternion [-1.000, 0.008, -0.003, 0.020]
            in RPY (radian) [-3.101, -0.005, -0.015]
            in RPY (degree) [-177.702, -0.303, -0.870]
At time 31.05, (current time 31.636)
- Translation: [0.104, -0.188, 0.079]
- Rotation: in Quaternion [-1.000, 0.010, -0.007, 0.019]
            in RPY (radian) [-3.103, -0.013, -0.020]
            in RPY (degree) [-177.796, -0.771, -1.147]
At time 32.05, (current time 32.636)
- Translation: [0.106, -0.186, 0.078]
- Rotation: in Quaternion [-1.000, 0.014, -0.006, 0.019]
            in RPY (radian) [-3.102, -0.011, -0.029]
            in RPY (degree) [-177.757, -0.621, -1.661]
At time 33.05, (current time 33.636)
- Translation: [0.106, -0.180, 0.078]
- Rotation: in Quaternion [-1.000, 0.012, 0.000, 0.013]
            in RPY (radian) [-3.116, 0.000, -0.024]
            in RPY (degree) [-178.505, 0.019, -1.392]
At time 34.05, (current time 34.636)
- Translation: [0.109, -0.174, 0.078]
- Rotation: in Quaternion [-1.000, 0.008, -0.005, 0.012]
            in RPY (radian) [-3.117, -0.010, -0.016]
            in RPY (degree) [-178.571, -0.547, -0.914]
At time 35.05, (current time 35.636)
- Translation: [0.110, -0.172, 0.078]
- Rotation: in Quaternion [-1.000, 0.009, -0.005, 0.016]
            in RPY (radian) [-3.110, -0.009, -0.019]
            in RPY (degree) [-178.208, -0.544, -1.074]
```