usnistgov/ARIAC #25: moving obstacles

dan9thsense, 2020-03-21:
Do the moving obstacles always start at the end of an aisle?  If so, is it always the same end (e.g., near the conveyor belt)?

zeidk, 2020-03-21:
They will spawn in any of the 4 aisles but will always spawn an one of the extremities of an aisle (never between the extremities). Also, there will be up to 2 moving obstacles in the environment.

I will make a big push before tomorrow with updates to the wiki.

zeid

dan9thsense, 2020-03-21:
Thanks for the quick response.  Looking more carefully at the sim, I see that the dynamic obstacles start moving before my call to start the environment.  The effect of that is the obstacles do "start" in the middle of the aisles since that may be where they are when the environment starts.

Is there a maximum time that they can have to traverse the aisle?

I am trying to figure out a good way to absolutely know at some point in time whether or not there is a dynamic obstacle present.

zeidk, 2020-03-22:
This happens when you have two moving obstacles spawning in the environment and not when you have only 1 moving obstacle (let me know if you see differently).
The idea is to have the second moving obstacle reach the opposite side even before the competition starts, and then both obstacles will move in sync in the opposite direction. Do you actually get the competition to start even before the obstacle reaches the other side? We may either ask you to wait until one of the 2 obstacles reaches the other side (which is not ideal) or we may change the script so they both start on the opposite ends. I will talk to the team tomorrow.

There is a maximum idle time and a maximum travel time. Just an idea but having 2 break beam sensors in an aisle can inform you on the speed of the obstacles and their directions.

Thank you for the inputs Dan.

dan9thsense, 2020-03-22:
Now I understand. Sounds good and thanks for the fast response.

closed 2020-03-22 by dan9thsense