usnistgov/ARIAC #26: updated after order completed

dan9thsense, 2020-03-21:
In qual_1_a, if the parts get delivered such that the second part needed for the updated order is delivered last in the original order, then the update is not triggered until after the original order is completed.  This means that we have to wait around a bit after completing an order before submitting it, just in case there is an updated order about to show up.

Is there some way to know how long we should wait?  In other words, is there some maximum time after an order is fully delivered to the tray that there will not be an updated order sent?

zeidk, 2020-03-22:
I think I should modify this so the update kicks in even before the order is complete, that would be an alternative.
The other alternative is to have a field in the order showing that this order will need to be updated at some point.

The first case seems to align more with a real situation. Let me know which one makes sense to you and I will discuss with the team tomorrow.

dan9thsense, 2020-03-22:
It's actually OK just to leave it as is, if it just means delaying submissions for a couple seconds. I did some testing and the update arrives within a second or so of the needed part being dropped, which is fine.  By the time the code gets from dropping the part to submitting the order, the new order generally has arrived and, if the code waits for 2 seconds to submit, then the order always gets there.

At this point, my suggestion is to not change it.

closed 2020-03-22 by dan9thsense

zeidk, 2020-03-22:
Sounds good. In this case I need to make it clear somewhere on the wiki that an update of the current order may come in seconds after the first order is complete.

dan9thsense, 2020-03-23:
It actually sends the order right after the part hits the tray.  If that part happens to complete the original order, then you may need to allow 2 seconds after the part hits the tray before submitting the order.  That seems to be plenty of time for the new order message to arrive.