usnistgov/ARIAC #27: updated order priority bonus?

dan9thsense, 2020-03-21:
Do updated orders not get the x3 priority order bonus?  Didn't happen for qual_1_a.

zeidk, 2020-03-22:
I will check the script on this one and will report back.

zeidk, 2020-03-26:
Hi Dan, are you talking about updated order or high priority order (interruption)? Did you get a chance to try qual_a_3.yaml where there is an order interruption?

dan9thsense, 2020-03-27:
I was asking if the updated orders get a 3x priority order bonus.  When a new order that is not an update comes in, that gets the 3x bonus as it should.  I was wondering if the updated orders also get a x3 bonus.

zeidk, 2020-03-28:
Updated orders will not get a priority bonus. We may implement this for next year but it will be less than x3 for sure.

dan9thsense, 2020-03-28:
OK, thanks for the clarification.

closed 2020-03-28 by dan9thsense