usnistgov/ARIAC #28: robot_description not found in docker run

dan9thsense, 2020-03-22:
When trying to run inside docker, the move group setup fails because it cannot find robot_description.  Is there a different namespace for that in the docker version?  Can you provide an example?

Modifying `run_team_system.bash`, replacing `rosrun ariac_example ariac_example_node` with `rosrun ariac_example moveit_example` results in this output:

```
[ INFO] [1584895281.650363526]: Loading gazebo_ros_control plugin
[ INFO] [1584895281.650491640]: Starting gazebo_ros_control plugin in namespace: /ariac/gantry
[ INFO] [1584895281.650881238]: gazebo_ros_control plugin is waiting for model URDF in parameter [/ariac/gantry/robot_description] on the ROS param server.
[ INFO] [1584895281.812392778]: Loaded gazebo_ros_control.
[Msg] VacuumGripper plugin loaded
[Msg] VacuumGripper plugin loaded
Error [parser_urdf.cc:3170] Unable to call parseURDF on robot model
Error [parser.cc:406] parse as old deprecated model file failed.
Error [parser_urdf.cc:3170] Unable to call parseURDF on robot model
Error [parser.cc:406] parse as old deprecated model file failed.
```
The sim starts but just eventually times out.

With my own code, it is unable to find robot_description.  I'm using the namespace `/ariac/gantry`, so the full patlh is `/ariac/gantry/robot_description`  It finishes quickly since it cannot assign the moveit groups:
```
[ERROR] [1584894762.678864949]: Robot semantic description not found. Did you forget to define or remap '/ariac/gantry/robot_description_semantic'?
[FATAL] [1584894762.784235012]: Group 'Full_Robot' was not found.
```

zeidk, 2020-03-22:
The Error you get with parse_urdf can be ignored and is deprecated. 
I will fix this so it works with moveit_example. It seems moveit is not properly loaded in the image.
I will try to get this to work before tomorrow.

zeid

dan9thsense, 2020-03-25:
Any luck with this?  We're getting pretty close to the qual runs and it takes time to get docker up and running well.

zeidk, 2020-03-25:
Will be pushed today Dan. Had some power outage around here.
Thank you,

zeid

zeidk, 2020-03-26:
Fixed it! I tried to push the new images to dockerhub but the website seems to be down or overloaded. Will try again tomorrow.

dan9thsense, 2020-03-26:
Thanks for the fast action.  I'll give it a try tomorrow.

zeidk, 2020-03-26:
I will let you know when it's ready. 

zeid

zeidk, 2020-03-28:
Dan, 
I have pushed the new images to dockerhub and modified docker scripts here https://github.com/usnistgov/ariac-docker. The docker container will start moveit automatically (your issue was due because moveit was not started). Some teams may not use moveit at all , if this is the case then I will have to make some modifications to provide moveit as an option.

- I have added `moveit_example_team` in `team_config` as another submission example. 
- The script `run_team_system.bash` will run a python script (`moveit_example_docker.py`) which is a simplified version of  `moveit_example.py`.  
  - `moveit_example.py` will not work in the competition mode because it calls the service `/ariac/material_locations`, which is forbidden in competition mode (see the [Cheats](https://github.com/usnistgov/ARIAC/blob/master/wiki/documentation/competition_interface_documentation.md) section).
- You can try the new script with:
  -  `./prepare_team_system.bash moveit_example_team`
  - `./run_trial.bash moveit_example_team sample_docker`

- Try the new docker images with your own code and let me know how it goes.
- I will update the wiki tomorrow to reflect those changes.

dan9thsense, 2020-03-28:
The good news is that it works, both with the example code and my code, both for single and all trials.
However, there are some issues:
It looks like the namespace for docker (ns="") is different from the namespace (ns="/ariac/gantry") when running outside of docker which means my launch file will need a parameter for ns.  Is that correct?  If so, no big deal, just want to be sure.

There are some differences in performance, perhaps related to not having the GUI running, and I want to see what's going on.  However, `roslaunch nist_gear gear_playback.launch` fails with the following errors:

```
[Err] [SystemPaths.cc:412] File or path does not exist["/home/ariac/ariac_ws/src/nist_gear/robots/torso/meshes/torso_base.dae"]
[Err] [MeshShape.cc:66] No mesh specified
[Err] [SystemPaths.cc:412] File or path does not exist["/home/ariac/ariac_ws/src/nist_gear/robots/torso/meshes/torso_main.dae"]
[Err] [MeshShape.cc:66] No mesh specified
[Err] [SystemPaths.cc:412] File or path does not exist["/home/ariac/ariac_ws/src/nist_gear/robots/ur10/ur_description/meshes/ur10/collision/base.stl"]
```

On my system, the correct paths are:
`~/ariac_ws/src/ARIAC/nist_gear/robots/torso/meshes/torso_base.dae`

and

`~/ariac_ws/src/ARIAC/nist_gear/robots/ur10/ur_description/meshes/ur10/collision/base.stl`

It is not obvious to me where those paths are being set or I would have tried changing them to get it to run.

zeidk, 2020-03-29:
I will try this today and in the best case scenario I will only need to change the path in gear_playback.launch

Thank you

zeidk, 2020-03-30:
Hi Dan,

Everything works fine on my end.
Do you have another nist_gear installed on your machine?
It seems the following line is pointing to `libLogPlaybackPlugin.so` located somewhere else (from a different `nist_gear package`).
`args="-p $(arg state_log_path) -s $(find nist_gear)/../../lib/libLogPlaybackPlugin.so --pause $(arg extra_gazebo_args)" required="true" />`
Can you try running everything in the `gear_playback.launch`  using the command line?

dan9thsense, 2020-03-31:
I pulled the code out of the launch file and ran it with absolute addresses and get the same result.
Here is the first command line:
`
rosrun gazebo_ros gzserver -p /home/dbarry/ariac_ws/ariac-docker/logs/sirius/march28/sample/gazebo/state.log -s /home/dbarry/ariac_ws/devel/lib/libLogPlaybackPlugin.so --pause --verbose __name:="gazebo"
rosrun gazebo_ros gzclient __name:="gazebo_gui" output="screen" respawn="false"`

and here is the second one (in another terminal):
rosrun gazebo_ros gzclient __name:="gazebo_gui" output="screen" respawn="false"

The output from the first one is:
```
Gazebo multi-robot simulator, version 9.0.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[ INFO] [1585627001.129092118]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1585627001.130171406]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.1.213
[Msg] 
Log playback:
  Log Version: 1.0
  Gazebo Version: 9.12.0
  Random Seed: 163728887
  Log Start Time: 0 2000000
  Log End Time: 86 332000000
World created. Model count: 81
[ INFO] [1585627002.566134902]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1585627002.593543098]: Physics dynamic reconfigure ready.
[Dbg] [ProximityRayPlugin.cc:74] Setting update rate of parent sensor to 20 Hz
[Dbg] [ProximityRayPlugin.cc:74] Setting update rate of parent sensor to 20 Hz
[Dbg] [ProximityRayPlugin.cc:74] Setting update rate of parent sensor to 20 Hz
[Dbg] [ProximityRayPlugin.cc:74] Setting update rate of parent sensor to 20 Hz
[Dbg] [ProximityRayPlugin.cc:74] Setting update rate of parent sensor to 20 Hz
[Dbg] [ProximityRayPlugin.cc:74] Setting update rate of parent sensor to 20 Hz
[Dbg] [ProximityRayPlugin.cc:74] Setting update rate of parent sensor to 20 Hz
[Dbg] [ProximityRayPlugin.cc:74] Setting update rate of parent sensor to 20 Hz
```

The output from the second one is:

```
[ INFO] [1585627052.535838817]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1585627052.537048504]: waitForService: Service [/gazebo_gui/set_physics_properties] has not been advertised, waiting...
[Err] [REST.cc:205] Error in REST request

libcurl: (51) SSL: no alternative certificate subject name matches target host name 'api.ignitionfuel.org'
```

At this point the gazebo window is open and the parts are in place, but there is no robot present.  When I hit the play button, that is when the errors (in the first terminal window) appear:

```
[Err] [SystemPaths.cc:412] File or path does not exist["/home/ariac/ariac_ws/src/nist_gear/robots/torso/meshes/torso_base.dae"]
[Err] [MeshShape.cc:66] No mesh specified
[Err] [SystemPaths.cc:412] File or path does not exist["/home/ariac/ariac_ws/src/nist_gear/robots/torso/meshes/torso_main.dae"]
[Err] [MeshShape.cc:66] No mesh specified
[Err] [SystemPaths.cc:412] File or path does not exist["/home/ariac/ariac_ws/src/nist_gear/robots/ur10/ur_description/meshes/ur10/collision/base.stl"]
[Err] [MeshShape.cc:66] No mesh specified
[Err] [SystemPaths.cc:412] File or path does not exist["/home/ariac/ariac_ws/src/nist_gear/robots/ur10/ur_description/meshes/ur10/collision/base.stl"]
[Err] [MeshShape.cc:66] No mesh specified
[Err] [SystemPaths.cc:412] File or path does not exist["/home/ariac/ariac_ws/src/nist_gear/robots/ur10/ur_description/meshes/ur10/collision/shoulder.stl"]
[Err] [MeshShape.cc:66] No mesh specified
[Err] [SystemPaths.cc:412] File or path does not exist["/home/ariac/ariac_ws/src/nist_gear/robots/ur10/ur_description/meshes/ur10/collision/upperarm.stl"]
[Err] [MeshShape.cc:66] No mesh specified
[Err] [SystemPaths.cc:412] File or path does not exist["/home/ariac/ariac_ws/src/nist_gear/robots/ur10/ur_description/meshes/ur10/collision/forearm.stl"]
[Err] [MeshShape.cc:66] No mesh specified
[Err] [SystemPaths.cc:412] File or path does not exist["/home/ariac/ariac_ws/src/nist_gear/robots/ur10/ur_description/meshes/ur10/collision/wrist1.stl"]
[Err] [MeshShape.cc:66] No mesh specified
[Err] [SystemPaths.cc:412] File or path does not exist["/home/ariac/ariac_ws/src/nist_gear/robots/ur10/ur_description/meshes/ur10/collision/wrist2.stl"]
[Err] [MeshShape.cc:66] No mesh specified
[Err] [SystemPaths.cc:412] File or path does not exist["/home/ariac/ariac_ws/src/nist_gear/robots/ur10/ur_description/meshes/ur10/collision/wrist3.stl"]
[Err] [MeshShape.cc:66] No mesh specified
[Err] [SystemPaths.cc:412] File or path does not exist["/home/ariac/ariac_ws/src/nist_gear/robots/ur10/ur_description/meshes/ur10/collision/shoulder.stl"]
[Err] [MeshShape.cc:66] No mesh specified
[Err] [SystemPaths.cc:412] File or path does not exist["/home/ariac/ariac_ws/src/nist_gear/robots/ur10/ur_description/meshes/ur10/collision/upperarm.stl"]
[Err] [MeshShape.cc:66] No mesh specified
[Err] [SystemPaths.cc:412] File or path does not exist["/home/ariac/ariac_ws/src/nist_gear/robots/ur10/ur_description/meshes/ur10/collision/forearm.stl"]
[Err] [MeshShape.cc:66] No mesh specified
[Err] [SystemPaths.cc:412] File or path does not exist["/home/ariac/ariac_ws/src/nist_gear/robots/ur10/ur_description/meshes/ur10/collision/wrist1.stl"]
[Err] [MeshShape.cc:66] No mesh specified
[Err] [SystemPaths.cc:412] File or path does not exist["/home/ariac/ariac_ws/src/nist_gear/robots/ur10/ur_description/meshes/ur10/collision/wrist2.stl"]
[... cut ...] 6] No mesh specified
[Err] [SystemPaths.cc:412] File or path does not exist["/home/ariac/ariac_ws/src/nist_gear/robots/ur10/ur_description/meshes/ur10/collision/wrist3.stl"]
[Err] [MeshShape.cc:66] No mesh specified
Segmentation fault
```

The second terminal does not have any messages, it just hangs at this point.

Note that there is no directory on this machine corresponding to `/home/ariac` much less all those subdirectories listed below it. 
I have no idea where it is getting that path name.

zeidk, 2020-03-31:
Everything looks good until you press the play button. Gazebo opening without showing the robot is expected and the robot appears and moves when you press the play button.
Let me look into this issue today.

zeid

zeidk, 2020-03-31:
Can you email me (zeid.kootbally@nist.gov) the file ~/.ariac/log/gazebo/state.log?

dan9thsense, 2020-04-01:
File sent just now.

zeidk, 2020-04-06:
I will look into this issue today.

zeidk, 2020-05-09:
@dan9thsense Finally figured out how to fix this and the instructions can be found [here](https://github.com/usnistgov/ARIAC/blob/master/wiki/tutorials/automated_evaluation.md#playing-back-the-simulation). Please give it a try and close the ticket if you are satisfied.

dan9thsense, 2020-05-10:
The state.log file is no longer showing up in the logs.  The folder that used to contain it, `gazebo`, is no longer present either.

zeidk, 2020-05-10:
@dan9thsense Can you check that you have the following at the top of your trial config file?
```
options:
   gazebo_state_logging: true
```

dan9thsense, 2020-05-10:
Ah, that fixed it.  I got it to work, thanks for figuring it out.  However, the instructions are not quite correct.  They say:

The easiest way to not get these errors is to create the following directory structure on the host

 `/home/ariac/ariac_ws/src`, then copy the whole robots directory from ARIAC into 

`/home/ariac/ariac_ws/src`

But what you actually need is to create the following directory structure on the host

 `/home/ariac/ariac_ws/src/nist_gear`, then copy the whole robots directory from ARIAC into 

`/home/ariac/ariac_ws/src/nist_gear`

Also, it is easy to make the mistake of adding the `ariac/ariac_ws....` directory to your home user directory, but actually you need to add it like a new user folder, directly under the `/home` directory.

closed 2020-05-10 by dan9thsense

zeidk, 2020-05-10:
oops, correct, I will fix this right away. 

Thanks!

stevegray954, 2020-05-11:
@zeidk Just saw your instructions -- instead of symlinking each file like you've got with your example, the easiest way without copying is
```bash
sudo mkdir -p /home/ariac/ariac_ws/src
sudo ln -s /home/YOUR_USER/ariac_ws/src/ARIAC/nist_gear /home/ariac/ariac_ws/src/
```