usnistgov/ARIAC #36: add a local folder?

dan9thsense, 2020-03-29:
To speed development with docker, I'd like to add a local folder to the docker image.  Can I do that with the build_team_system.bash script?

With dockerfiles there is the `add` command:

add ~/some_local_folder ~/my_team_ws/src

but the add command does not work with bash

zeidk, 2020-03-30:
Not sure what you are trying to do with the local folder but for testing purposes you can look at how it is done in `build_team_system.bash`
Everything in `build_team_system.bash` is happening in the image, so `mkdir ~/some_local_folder` will actually create `some_local_folder` in your image.

dan9thsense, 2020-03-30:
What I want to do is copy a file from my local drive into the image, rather than using wget or cloning a repository.  That way I can test changes to the coffee without having to update the repo for every little change.

Pavel-P, 2020-04-01:
For local testing you can use docker cp (https://docs.docker.com/engine/reference/commandline/cp/).
This is run outside of the container though, so it won't work in `build_team_system.bash`

dan9thsense, 2020-04-02:
Hmmm, I was hoping there would be a way to integrate it into the build_team_system.bash or run_team_system.bash files, so it would work seamlessly.  Seems odd that it can pull a repo but not pull from the local drive.

stevegray954, 2020-04-02:
I'd like to second this request.

`build_team_system.bash` calls `build_competitor_image.bash`

@dan9thsense You'd have to copy your file/folder into the docker build context (the `ariac-docker/ariac-competitor` folder.  The [bash script](https://github.com/usnistgov/ariac-docker/blob/master/ariac-competitor/build_competitor_image.bash) already copies the build and run team system scripts there for this reason. 

Inside the `Dockerfile`, why not copy the entire directory instead just those two files one at a time?

zeidk, 2020-04-05:
After some testing, I think I found a way to do just that but there is a caveat.

1. You need to modify [ariac-docker/ariac-competitor/ariac-competitor/Dockerfile](https://github.com/usnistgov/ariac-docker/blob/master/ariac-competitor/ariac-competitor/Dockerfile)
2. Copy the following command in `ariac-docker/ariac-competitor/ariac-competitor/Dockerfile` to copy the directory `ariac_ws` (from your computer) to `/home/ariac-user/ariac_ws` (in your image): `COPY ariac_ws /home/ariac-user/ariac_ws`
- Here is the caveat, the first argument of the `COPY` command (`ariac_ws`) should be in your context, i.e., it should also be in the directory `ariac-docker/ariac-competitor/ariac-competitor`. There are some ways to copy files and directories from outside the context but I will need to make some modifications and also long testing...and I will break things.
- It's probably a good idea to paste the above command right after `FROM zeidknist/ariac4-competitor-base-melodic:latest` in `Dockerfile`
3. Modify `build_team_system.bash`:
- Remove commands that create a workspace and fetch the zipp file.
-  `cd` into the workspace (`/home/ariac-user/ariac_ws`) and compile it.

zeidk, 2020-05-06:
I will close this for now. Please follow the instructions above and let me know if I need to reopen this ticket.

closed 2020-05-06 by zeidk