usnistgov/ARIAC #37: Conveyor belt active in runs where it shouldn't be active

stevegray954, 2020-03-30:
Running without the `--development-mode` flag, `ariac/conveyor/state` is always `power: 100, enabled: True` when `/ariac/start_competition` is called. This is true for all the `qual_a` scenarios, including 1 and 3 - though in their descriptions they say `Conveyor belt is not active.`

zeidk, 2020-03-30:
https://github.com/usnistgov/ARIAC/blob/787cf273a81294d57e0e8fa19df3f9636ebf5653/nist_gear/src/ROSConveyorBeltPlugin.cc#L112
It seems in ARIAC 2019 the belt was forced to be enabled in competition mode (even though it was set to "disabled" in the config file). 
Do you remember this happening last year and the reasoning behind this? I can contact Shane from OSRF and learn more about this behavior. I will also talk to the NIST team.

stevegray954, 2020-03-30:
I didn't end up using the belt last year, but it is very likely that it did behave the same way.

It looks like the task manager always calls the following on startup, but it could be made conditional based on `belt_population_cycles`
https://github.com/usnistgov/ARIAC/blob/787cf273a81294d57e0e8fa19df3f9636ebf5653/nist_gear/src/ROSAriacTaskManagerPlugin.cc#L653

I also noticed the `belt_population_cycles` param doesn't seem to be working -- in `qual_a_2` it's set to 3, but the parts only appear once.

stevegray954, 2021-02-07:
OBE

closed 2021-02-07 by stevegray954