usnistgov/ARIAC #38: qual_a_3 is missing pulley parts

stevegray954, 2020-04-03:
From the description, `qual_a_3` should have a pulley to be flipped - and the order is looking for a red pulley.

However, there are no pulleys - only gears and gaskets.  I think one of the blue gear areas - either on the shelf or bin - should be a red pulley.

stevegray954, 2020-04-03:
Check out PR #39 -- I changed the blue gears in the bin to be red pulleys.

zeidk, 2020-04-03:
Thanks, will take care of some issues today including this one.

closed 2020-04-05 by zeidk

dan9thsense, 2020-04-13:
I just saw this today.  Please do not change the qual files.  I saw that there were no pulleys in qual_a_3 and assumed that it was an example of "not enough parts."  I already submitted code for the qualifiers (within the original April 6th to 10th window) and do not have time right now to revise it.  I know I sound like a broken record here, repeating myself, but please don't make late changes.

stevegray954, 2020-04-13:
I had updated qual_a_3 to match the description [here:](https://github.com/usnistgov/ARIAC/blob/master/wiki/qualifiers/qualifier_scenarios.md#part-a-trial-3)
The description says that a pulley needs to be flipped, which made it odd there were no pulleys.
I'm fine with reverting, but then the description should change.

zeidk, 2020-04-13:
Dan, your code will be tested against **qual_b** files which you do not have access. **qual_a** files are meant for participants to try their system and whatever you submitted will not be tested using any **qual_a** files. As long as your system deals with flipped parts then you should be good.

dan9thsense, 2020-04-13:
Oh, i thought the quals used both the qual_a and the qual_b scenarios to generate a score for 6 trials.  It only uses qual_b?

stevegray954, 2020-04-13:
Agreed, in previous years, the quals used both 'a' and 'b' scenarios.

dan9thsense, 2020-04-18:
Can we get an answer to this question (Is qual_a used for the qualifiers?) ?  It's pretty important for the qualifiers.