usnistgov/ARIAC #40: Missing RVIZ folder in Nist_gear

frank-qcd-qk, 2020-04-10:
Observed following error during catkin build:
------------------------------------------------------------------------------------------------------------------------------------
Errors     << nist_gear:install /home/frank/ariac_ws/logs/nist_gear/build.install.000.log                            
CMake Error at cmake_install.cmake:675 (file):
  file INSTALL cannot find
  "/home/frank/ariac_ws/src/ARIAC_ORIGINAL/nist_gear/rviz".

make: *** [install] Error 1
cd /home/frank/ariac_ws/build/nist_gear; catkin build --get-env nist_gear | catkin env -si  /usr/bin/make install; cd -
------------------------------------------------------------------------------------------------------------------------------------
It seems that there is a missing folder?

frank-qcd-qk, 2020-04-10:
@zeidk

zeidk, 2020-04-14:
I removed vendor and rviz directory from install (from ARIAC 2019). Also temporarily disabled test_ariac_scorer until we fix the scoring.

closed 2020-04-14 by zeidk

zeidk, 2020-04-14:
Also note that running catkin with install will not install some scripts. Modify the CMakeLists.txt to install what you need. This will be fixed before after the qualifiers.