usnistgov/ARIAC #41: Parts auto-magically pop-off from AGV

avidacs, 2020-04-11:
When running qual_a_2, some parts are suddenly dropped from the agv tray while the robot is picking up part somewhere else. Part dropped from agv2 tray: gasket_part_red, The config file includes: 
---
drop_regions:
    shipping_box_0_impeding:
      frame: agv1::kit_tray_1
      min:
        xyz: [-0.3, -0.3, 0.0]
      max:
        xyz: [0.3, 0.3, 0.5]
      destination:
        xyz: [0.3, 0.3, 0.05]
        rpy: [0, 0, 0.2]
      product_type_to_drop: gasket_part_red
---
But this drop happens on AGV2 and not from gripper but after the part already sitting on tray for a while. Is that behavior supposed to be normal?

I try to add a small video of the event
[auto-amgical-drop.zip](https://github.com/usnistgov/ARIAC/files/4464165/auto-amgical-drop.zip)

zeidk, 2020-04-13:
Thank you, I will try to reproduce this issue. When submitting AGV2 do you still get points for parts that were popped off?

avidacs, 2020-04-13:
Nope, no points for the dropped-off products.

stevegray954, 2020-04-13:
Ha, that's a fun one.  Interesting that the parts move exactly when you pick up a new blue gear.  Looks like both red gaskets and blue gears are supposed to drop in the full version of the drop section: 

```
  drop_regions:
    shipping_box_0_impeding:
      frame: agv1::kit_tray_1
      min:
        xyz: [-0.3, -0.3, 0.0]
      max:
        xyz: [0.3, 0.3, 0.5]
      destination:
        xyz: [0.3, 0.3, 0.05]
        rpy: [0, 0, 0.2]
      product_type_to_drop: gasket_part_red
    shipping_box_1_impeding:
      frame: agv2::kit_tray_2
      min:
        xyz: [-0.3, -0.3, 0.0]
      max:
        xyz: [0.3, 0.3, 0.5]
      destination:
        xyz: [0.3, 0.3, 0.05]
        rpy: [0, 0, 0.2]
      product_type_to_drop: gear_part_blue
```

stevegray954, 2020-04-13:
Edit: This actually makes a ton of sense.  

Quick fix:
Need to get rid of this part of the config file.  It's telling the red gaskets that they have a drop scheduled (see https://github.com/usnistgov/ARIAC/blob/ff420d41917defa9fdf84ae1f24c8950fa0959d2/nist_gear/src/VacuumGripperPlugin.cc#L602))

```
    shipping_box_0_impeding:
      frame: agv1::kit_tray_1
      min:
        xyz: [-0.3, -0.3, 0.0]
      max:
        xyz: [0.3, 0.3, 0.5]
      destination:
        xyz: [0.3, 0.3, 0.05]
        rpy: [0, 0, 0.2]
      product_type_to_drop: gasket_part_red
```

However, this doesn't fix the underlying issue.

The simulation has changed -- never before could the arm pick up a part and then not have it be dropped before picking up another part.  The red gasket goes to AGV2, so it never encounters AGV1 and gets dropped.  `dropPending` remains true.

Right here at https://github.com/usnistgov/ARIAC/blob/ff420d41917defa9fdf84ae1f24c8950fa0959d2/nist_gear/src/VacuumGripperPlugin.cc#L460, it just checks if any part is attached to the gripper and a drop is pending.  However, this line https://github.com/usnistgov/ARIAC/blob/ff420d41917defa9fdf84ae1f24c8950fa0959d2/nist_gear/src/VacuumGripperPlugin.cc#L464 should catch that the attached object is the wrong type.

zeidk, 2020-04-13:
Good catch, I will take care of this ASAP. Again, thank you Steve for digging into this issue.

zeidk, 2020-04-13:
@avidacs I assume 1 gear_blue_part drops when you try to place it in AGV2, correct? While I take care of the source code, can you, for now, change `frame: agv1::kit_tray_1` to `frame: agv2::kit_tray_2` for `gasket_part_red` ?

Thanks

stevegray954, 2020-04-13:
Please don't forget to change the drop location too.  It's supposed to drop in the tray, not onto the ground.

stevegray954, 2020-04-13:
> Good catch, I will take care of this ASAP. Again, thank you Steve for digging into this issue. Also, the code wasn't changed at all, it's just it never took care of handling this kind of issue if the wrong AGV is issued in the config file.

Not only that, it wasn't designed to handle the robot being able to pick up and hold parts - it was specifically assuming you'd pick up the part then take it to its destination.

avidacs, 2020-04-14:
@stevegray954 No, the red gasket didn't drop from the gripper above tray - I was putting it onto agv2 by intentionally releasing it from the gripper (if that's what you meant). @zeidk Yes, the blue gears were dropped from the gripper above tray 2. Thanks for helping!

stevegray954, 2020-04-14:
Pretty sure I know how to fix this in the plugin. Want me to take a crack at it and push up a PR?  I won't be able to do so until tonight.

zeidk, 2020-04-14:
@stevegray954 Thank you Steve. I'm looking at the issue with the scoring. We made changes about the scoring and for some reasons they were never pushed. It happened when we had to leave NIST because of COVID-19. It's something easy to redo though but I need to test it before pushing the edits.
@Pavel-P do you have the necessary bandwidth to look at this current issue (#41) today? If not, Steve is willing to take a look.

Thank you

stevegray954, 2020-04-14:
With all these bugs being fixed, what happens with the rolling qualification window?  Are people being evaluated against different versions of the sim code?

Pavel-P, 2020-04-14:
I tried replicating this scenario on my end but my current computer's too slow to do much ariac debugging right now. My initial hunch was that moving this line: https://github.com/usnistgov/ARIAC/blob/ff420d41917defa9fdf84ae1f24c8950fa0959d2/nist_gear/src/VacuumGripperPlugin.cc#L504 into the `HandleDetach()` method would do the trick, but this seems to have some strange side effects.

Pavel-P, 2020-04-14:
As far as which code version we'll be running, I think the best option would be to let teams specify if they want to use a particular commit as long as that commit falls within the qualifier window. I don't think the version differences will be enough to change who does or doesn't qualify, and this way teams that already submitted aren't forced to make any new changes. We're open to other options though.

stevegray954, 2020-04-15:
@Pavel-P @zeidk I found the issue. Take a look at PR #46 

`dropPending` and `dropAttachedModel` were never cleared when a new part is picked up -- this is how the red pieces were still getting dropped when a new blue piece was picked up.  See this code block:
https://github.com/usnistgov/ARIAC/blob/ff420d41917defa9fdf84ae1f24c8950fa0959d2/nist_gear/src/VacuumGripperPlugin.cc#L617

The fix was to just clear `dropPending` and `dropAttachedModel` by default each time something new is picked up - then set them later in that function if needed.

Also, nothing checked that `dropAttachedModel` was matched the type of object in the hand -- the only check was that the object in hand was of a type that should be dropped.  See:
https://github.com/usnistgov/ARIAC/blob/ff420d41917defa9fdf84ae1f24c8950fa0959d2/nist_gear/src/VacuumGripperPlugin.cc#L464

@avidacs I tested this against my system, but could you go ahead and try the pull request, too?

stevegray954, 2020-04-15:
@Pavel-P @zeidk PR #47 also includes fixes for `qual_a_2.yaml` -- part drop over the correct agv and  they now drop over the tray instead of landing on the ground.

avidacs, 2020-04-15:
@stevegray954 I pulled and tried, works as expected :) Thanks a lot!

zeidk, 2020-04-15:
@stevegray954 Just tried it myself and the issue is gone about parts popping off. Good job Steve and thank you.

After your suggestion about fixing the part drop areas I discussed with the team  about parts dropping on the floor when the robot places them in the tray (which is implemented in qual_a_2). Although in previous ARIAC parts would drop only in trays, ARIAC 2020 includes parts falling on both the floor and in kit trays. From the [wiki](https://github.com/usnistgov/ARIAC/blob/master/wiki/documentation/agility_challenges.md): "Recovery could include retrieving the dropped product or fetching a new product".  If necessary, I will modify this page to make it explicit that placing parts can include both.

Participants should have a plan to deal with both situations.  I will merge the pull request on the qual_2_a fix but the finals will include both these situations while parts are placed in trays. After the qualifiers, I will create a yaml file which specifically includes both situations.

Thanks again for your help :)

stevegray954, 2020-04-18:
@zeidk @Pavel-P @anthonydowns We may want to change the way the part dropping works, then, if the parts can't always be recovered.  Right now, each arm tracks what parts it has dropped - so if a person has to move two instances of a blue gear, they would drop one if they used the same arm to move both, but would drop two if they used both arms.  

Alternatively, maybe just make sure there are never insufficient numbers of the parts that will be dropped such that dropping 1 vs 2 impacts a competitors ability to get all products?

avidacs, 2020-04-20:
I can confirm this: "...so if a person has to move two instances of a blue gear, they would drop one if they used the same arm to move both, but would drop two if they used both arms". Our solution uses two arms so the first _two_ gear parts are both dropped the same way one after the other. (And that is kind of strange...)

zeidk, 2020-04-21:
This seems like a bug rather than being the intended behavior. 
**The intended behavior**: Drop only the first blue gear that is being placed in the tray.
**The actual behavior**: Drop only the first blue gear that each arm is placing in the tray.

We will need to get this fixed as soon as the qualifiers end. Thanks for reporting this.

stevegray954, 2020-04-21:
@zeidk Yeah, it's really an artifact of the competition history.  Each arm runs an instance of the plugin, which was great when there was only one arm in the beginning.

Made a new issue #53

stevegray954, 2021-02-07:
Initial issue resolved by PR #46 and #47.

Remaining issue still exists, but is captured in #53

Someone please close this.

zeidk, 2021-05-02:
Fixed.

closed 2021-05-02 by zeidk