usnistgov/ARIAC #43: Tag version of code for qualifier

stevegray954, 2020-04-13:
Related to @dan9thsense's comment in _https://github.com/usnistgov/ARIAC/issues/38#issuecomment-612939064_

It would be good to tag a version of the code we know is going to be used for qualification evaluation, while development continues on code to be used in the finals.

zeidk, 2020-04-17:
I think I will lock development for qualifiers today to start testing submissions. The most recent changes we made (yours included) are in favor of participants and I can't think of anything we recently changed that would disadvantage anyone. Further bug fixes between now and the start of finals will be done in a separate branch and will be merged after the qualifiers.

To answer your most recent github question about docker. I did some testing today with automating the creation of docker images based on specific ARIAC commits. It takes too long (1h per team). We have lots of teams this year (I dont think I'm allowed to provide the number) and we are running everything on slower machines since we left NIST. I would like to go ahead and use the current ARIAC version as the official version for the qualifiers.

stevegray954, 2020-04-17:
@zeidk Please let us know when the new Docker image is available so we can start testing with it.  Thanks!

zeidk, 2020-04-17:
Yes will do

dan9thsense, 2020-04-17:
The recent scoring change can certainly relatively reduce a team's score compared to other teams.  Code submitted before this change will not deliver parts with wrong colors.

stevegray954, 2020-04-18:
@zeidk @Pavel-P `zeidknist/ariac4-server-melodic` is 3 weeks old.  When will you post the new Docker image so we can test with Docker before submission?

zeidk, 2020-04-20:
Dan, a new image will be uploaded today. I am also trying to fix the playback not working properly. If I can't get the playback to work I will still upload the new image and will fix the playback afterwards. In that case you will be able to test with the new image. We will try to make a general announcement on the new image by email but conveying information to such a big audience requires some authorizations from NIST. So I will post here when the image is ready and also on the wiki page.

dan9thsense, 2020-04-20:
@zeidk I think you meant to direct your comment to @stevegray954

stevegray954, 2020-04-21:
@zeidk  I'd just like more detail on what's going on with qualifiers.  Everyone will be tested against the same version (using the same base docker image), right?  And that base docker image has yet to be released for competitors to test before submitting our qualifier code, which is due Friday?

zeidk, 2020-04-21:
@stevegray954 Yes, we will test your code with the new docker image (which is now pushed on dockerhub). The new docker scripts have also been updated here https://github.com/usnistgov/ariac-docker. 
[Playback](https://github.com/usnistgov/ARIAC/blob/master/nist_gear/launch/gear_playback.launch) will not work (it will crash if you are trying to see the result in gazebo from state.log)  but I'm still working on it. 

Don't forget to get the docker scripts first before you run `pull_docker_images.bash`
Below is the result of `moveit_example_team` using the new docker images.
```
Score breakdown:
<game_score>
	Total game score: [7]
	Total process time: [70.533]
	Arms collision?: [0]
<order_score order_0>
	Total order score: [7]
	Completion score: [7]
	Time taken: [70.53]
	Complete: [true]
	Priority: [1]
<shipment_score >
	Completion score: [7]
	Complete: [true]
	Submitted: [true]
	Product type presence score: [2]
	Product color presence score: [2]
	All products bonus: [2]
	Product pose score: [1]
	Delivered to correct agv: [true]
</shipment_score>

</order_score>

</game_score>
```

stevegray954, 2021-02-07:
OBE

closed 2021-02-07 by stevegray954