usnistgov/ARIAC #44: Product type presence point not counted when color is wrong

avidacs, 2020-04-14:
According to the updated scoring wiki page: 1 point is given for each correct product type touching the tray, and additional 1 point for each correct product color. However, it seems that no point is given when the color is wrong.

Example order:
Type: [piston_rod_part_blue]
Type: [gear_part_green]
Type: [gear_part_red]

Served parts:
kit_tray_1_bin4|piston_rod_part_red_1
kit_tray_1_bin3|gear_part_red_8
kit_tray_1_bin3|gear_part_red_6

Scoring:
<shipment_score >
	Completion score: [2]
	Complete: [false]
	Submitted: [true]
	Product presence score: [1]
	All products bonus: [0]
	Product pose score: [1]
	Delivered to correct agv: [true]
</shipment_score>

stevegray954, 2020-04-14:
Oof, yeah, the scoring wiki page seems wrong.

As far as the sim knows, the product type is a string which includes the color -- e.g. `piston_part_red`

Looking at the commits, no one has changed the (scoring code)[https://github.com/usnistgov/ARIAC/blob/master/nist_gear/src/AriacScorer.cpp] in over two months.  

@zeidk Why did the scoring description on the wiki change without changing the scoring code to match?

zeidk, 2020-04-15:
This has been fixed. Will push later today after running some trials myself.

zeidk, 2020-04-16:
The changes I made to the scoring system:
 - +1 pt awarded for correct part type.
 - +1 pt awarded for correct color.

To quickly test if the scoring works, spawn **Actual** parts in a tray using the command line. For instance, the 4 commands below  will spawn 4 parts in agv 1. You should only change model names if you want to  test with different parts.

Modify a trial config file with some custom orders, then read the order and submit the agv in your code.

```
rosrun gazebo_ros spawn_model -sdf -x -0.021891 -y 7.338831 -z 0.74 -R 0.000005 -P -0.000026 -Y 1.644586 -file `rospack find nist_gear`/models/piston_rod_part_blue_ariac/model.sdf -model piston_rod_part_blue_1
rosrun gazebo_ros spawn_model -sdf -x -0.016383 -y 7.238831 -z 0.74 -R 0.000005 -P -0.000026 -Y 1.644586 -file `rospack find nist_gear`/models/piston_rod_part_red_ariac/model.sdf -model piston_rod_part_red_2
rosrun gazebo_ros spawn_model -sdf -x -0.009871 -y 7.138831 -z 0.74 -R 0.000005 -P -0.000026 -Y 1.644586 -file `rospack find nist_gear`/models/piston_rod_part_green_ariac/model.sdf -model piston_rod_part_green_3
rosrun gazebo_ros spawn_model -sdf -x -0.002934 -y 7.038831 -z 0.74 -R 0.000005 -P -0.000026 -Y 1.644586 -file `rospack find nist_gear`/models/piston_rod_part_green_ariac/model.sdf -model piston_rod_part_green_4
```

In the table below, the **Required** field consists of parts required in the order (yaml file) while the **Actual** field consists of parts in the tray when the order is submitted.

I tested the changes in the scoring source files with the following configurations and the results are reported in the table below.

**CT**: Correct Type
**CC**: Correct Color

<table>
   <tr>
     <th>Required</th>
     <th>Actual</th>
     <th>CT</th>
     <th>CC</th>
   </tr>
   <tr>
     <td width="35%">piston_rod_part_green</td>
     <td width="35%">piston_rod_part_green</td>
     <td width="15%">1 pt</td>
<td width="15%">1 pt</td>
   </tr>
<tr>
     <td width="35%">gear_part_green</td>
     <td width="35%">gear_part_green</td>
     <td width="15%">1 pt</td>
<td width="15%">1 pt</td>
   </tr>
<tr>
     <td width="35%">gear_part_green</td>
     <td width="35%"></td>
     <td width="15%">0 pt</td>
<td width="15%">0 pt</td>
   </tr>
</table>

<table>
   <tr>
     <th>Required</th>
     <th>Actual</th>
     <th>CT</th>
     <th>CC</th>
   </tr>
   <tr>
     <td width="35%">piston_rod_part_green</td>
     <td width="35%">piston_rod_part_green</td>
     <td width="15%">1 pt</td>
<td width="15%">1 pt</td>
   </tr>
<tr>
     <td width="35%">gear_part_green</td>
     <td width="35%">gear_part_red</td>
     <td width="15%">1 pt</td>
<td width="15%">0 pt</td>
   </tr>
<tr>
     <td width="35%">gear_part_green</td>
     <td width="35%">gear_part_red</td>
     <td width="15%">1 pt</td>
<td width="15%">0 pt</td>
   </tr>
</table>

<table>
   <tr>
     <th>Required</th>
     <th>Actual</th>
     <th>CT</th>
     <th>CC</th>
   </tr>
   <tr>
     <td width="35%">gear_part_green</td>
     <td width="35%">gear_part_red</td>
     <td width="15%">1 pt</td>
<td width="15%">0 pt</td>
   </tr>
<tr>
     <td width="35%"></td>
     <td width="35%">gear_part_red</td>
     <td width="15%">0 pt</td>
<td width="15%">0 pt</td>
   </tr>
</table>

<table>
   <tr>
     <th>Required</th>
     <th>Actual</th>
     <th>CT</th>
     <th>CC</th>
   </tr>
   <tr>
     <td width="35%">piston_rod_part_green</td>
     <td width="35%">piston_rod_part_green</td>
     <td width="15%">1 pt</td>
<td width="15%">1 pt</td>
   </tr>
<tr>
     <td width="35%">gear_part_green</td>
     <td width="35%">gear_part_red</td>
     <td width="15%">1 pt</td>
<td width="15%">0 pt</td>
   </tr>
<tr>
     <td width="35%">gear_part_green</td>
     <td width="35%">gear_part_green</td>
     <td width="15%">1 pt</td>
<td width="15%">1 pt</td>
   </tr>
</table>

dan9thsense, 2020-04-16:
Wait, what?  The scoring is changing?

zeidk, 2020-04-17:
- Bonus for all correct parts stay the same.
- Total score is not doubled: total score takes into account other metrics (e.g., bonus, correct poses).
- Changes made:
  - Before, if `gear_part_blue` was required and you provided `gear_part_green` you got 0 pt. Now you get 1 pt because you provided at least the correct type (even if the color is off).
  - Before, if `gear_part_blue` was required and you provided `gear_part_blue` you got 1 pt. Now you get 1 pt for correct type (`gear_part`) + 1 pt for correct color (`blue`) = 2pts.

stevegray954, 2020-04-17:
@zeidk I'm assuming that correct type, but wrong color does not get the all product bonus.  Is that correct?

anthonydowns, 2020-04-17:
@stevegray954 , that is the intended scoring behavior.  
In order to get the All Product Bonus, you should have all the right part types, in the right position & orientation, of the right color, then you would get the all product bonus. 

@zeidk , is that how it is implemented currently?

zeidk, 2020-04-17:
@anthonydowns yes this is how it is currently implemented.

@stevegray954 Yes, will do.

dan9thsense, 2020-04-17:
It's a little late in the competition to be changing the scoring

dan9thsense, 2020-04-18:
@zeidk @stevegray954 The current all parts bonus (same in previous years) does not require the correct position or orientation.  It requires that all the parts are present, without any extra or faulty parts present.

zeidk, 2020-04-18:
Correct, I didn't catch that Steve mentioned correct poses. Parts with correct poses have their own set of points and are not part of the bonus for all parts.

anthonydowns, 2020-04-19:
OK, if that is how it has been done in previous years, it can stay that way for the qualifiers at least.  We (at NIST) will discuss if that will change for the Finals and let you all know.  This is not how I had intended the scoring to work, but as I said, it will stay that way for the qualifiers at least. Thanks for correcting this Dan & Steve.

dan9thsense, 2020-04-19:
what can stay?  The bonus or the part colors or both?

anthonydowns, 2020-04-20:
Sorry @dan9thsense , That wasn't clear on my part.  What I was intending to say was that the correct poses will continue to not be counted as a requirement for the all parts bonus, keeping this as it has been in the past.  This may change for the finals, but we'll let you know for sure after the Quals are done

anthonydowns, 2020-04-21:
ok, so for the qualifiers here is what the scoring will be for the parts:

For a kit containing _n_ parts:
Each correct part that is put into the kit, 1 point up to a max of _n_
Each correct colored part that is put into the kit, 1 point up to a max of _n_
if all the correct parts are in the kit and correctly colored, you get the all parts bonus (an additional _n_ points)

The correct position/orientation points are being treated as a separate pool of points, up to a max of _n_ .

If this changes for the Finals this year, we will make sure to let everyone know.  Thank you all

dan9thsense, 2020-04-22:
I'd like to convey the complexity of making code changes to conform to the new scoring rule that allows parts with incorrect colors to gain a score.  At first glance, this seems simple-- just search for parts with all colors and, while preferring ones with the correct colors, accept other colors as a secondary choice.  But it is much more complex than that.  Have you considered the decisions that need to be made when an order is changed or a priority order is issued?  The current tray contents are evaluated to see: 1) how close the current order is to completion; 2) how many parts on the tray can be used for the new order; 3) how many parts on the tray need to be removed for the new order; 4) how many available parts are there overall (this number is dynamic due to the conveyor belt);  5) how far away are the parts that need to be added to either a) complete the current order or b) use this tray to fulfill the new order and then come back later to fill the original order.  Considering all that, the code needs to find an optimal solution for overall time and parts, considering bonus points for priority orders and bonus points possibly lost due to not having all the correct parts (or colors, now).  As you might imagine, that code is pretty complex.  In my case, there is a lot of Monte Carlo evaluation used to find the best parameters to use for making the decision of whether to swap to the priority order right away or to finish the original order first.  The Monte Carlo aspects alone can take a week of 24/7 computer time to run through all the cases.   Allowing off-color parts to count toward the score nullifies all of that previous evaluation.  Furthermore, it requires code changes in complex areas, which means that once the code is revised to take the new rules into account, it then has to go back through all of the unit testing and debugging steps required to uncover the edge cases that could cause an inadvertent crash and a resultant zero score.

For this scoring change to show up in the last week before quals (and only a few weeks before finals) in a competition that has been going on for months is extremely frustrating, especially when one budgeted time to get it all working and submitted early in the quals window, knowing that there are other obligations at the end of April.

closed 2020-05-06 by zeidk