usnistgov/ARIAC #45: Topic Mapping for Arms?

mcboyd-bu, 2020-04-14:
The tutorials and samples for 2020 list arm-related topics as "/ariac/arm1..." and "/ariac/arm2/..." but when I run the environment with "roslaunch nist_gear sample_environment.launch" there are no "arm1" or "arm2" topics. There are "/ariac/gantry/left_arm_..." and "/ariac/gantry/right_arm_..." topics. Is there a mapping of arm arm1/2 to arm right/left I should know about? 
Accordingly, the ["Interacting with GEAR" tutorial](https://github.com/usnistgov/ARIAC/blob/master/wiki/tutorials/gear_interface.md) has many steps that don't work since the topics don't exist. For example, nothing in the ["Controlling Arm Joints" section](https://github.com/usnistgov/ARIAC/blob/master/wiki/tutorials/gear_interface.md#controlling-arm-joints) works as-is. I can spend the time figuring it all out, but I am hoping there is a mapping or an updated tutorial (and sample files) somewhere. Thank you!

zeidk, 2020-04-15:
This is correct, the top of the page has a description about the page being under construction and code on the page is for the previous ariac. 
@Pavel-P Do you think you can provide some examples of mapping for the gantry?

zeidk, 2020-04-15:
If this helps, the Actuators section on [this page](https://github.com/usnistgov/ARIAC/blob/master/wiki/documentation/competition_interface_documentation.md) covers robot's topics

closed 2020-05-06 by zeidk