usnistgov/ARIAC #49: collisions penalty

dan9thsense, 2020-04-16:
I know that the order score is zero if the robot contacts itself or a dynamic obstacle.  However, there does not seem to be any penalty for contacting structures such as shelves or the walls of the conveyor belt or the AGVs.  Is that correct behavior?

zeidk, 2020-04-16:
Yep, this is correct. The wiki on [scoring](https://github.com/usnistgov/ARIAC/blob/master/wiki/documentation/scoring.md) (see **COL** under **Performance metrics**) states that you will get a 0 if the arms collide with each other, the torso, or a moving obstacle. There is no penalty this year if the arms collide with anything else.

dan9thsense, 2020-04-16:
OK, thanks for the fast reply.

closed 2020-04-16 by dan9thsense