usnistgov/ARIAC #52: Simulation speedup for parts on robot torso tray

stevegray954, 2020-04-19:
Issue: adding parts onto the torso tray slows the sim to a crawl 
(from 0.98x -> 0.28x with 4 parts for my laptop with i7-8850H, 32GB RAM)

Suggestion: it'd be good to handle parts on the robot torso tray the way they're handled when attached to the gripper -- have gazebo create a temporary joint.  This joint would be removed when something other than the robot torso tray comes in contact with the part again.

avidacs, 2020-04-20:
I guess then that with this temporary joint the part slipping problem is also avoided, since all parts on the tray would be firmly attached to the robot. If that is the case, this change would have serious consequences on the chosen strategy how parts are to be handled by the robot. On the other hand, I'd be happy with any performance increase of the simulation :)

stevegray954, 2021-02-07:
Suggestion is still valid for 2021 as the gantry robot again has a tray.

zeidk, 2021-04-26:
This was implemented a while ago but forgot to close this ticket. Although you need to call the service yourself to attach/detach parts on the tray, joints are now created/deleted between parts and the gantry tray.

closed 2021-04-26 by zeidk