usnistgov/ARIAC #53: Arms track dropped parts separately

stevegray954, 2020-04-21:
@stevegray954 @avidacs This seems like a bug rather than being the intended behavior. 
**The intended behavior**: Drop only the first blue gear that is being placed in the tray.
**The actual behavior**: Drop only the first blue gear that each arm is placing in the tray.

We will need to get this fixed as soon as the qualifiers end. Thanks for reporting this.

_Originally posted by @zeidk in https://github.com/usnistgov/ARIAC/issues/41#issuecomment-617220330_

zeidk, 2020-05-06:
The plugin that drops parts is associated to each arm, as you mentioned. Rewriting the plugin to link it to the robot instead of the arms will take some time and I don't think I will be able to complete it before the start of the finals. I will still keep this ticket open so I can work on it after the finals (for ARIAC 2021).

In the mean time, I will rewrite the config files so that more than 2 parts of the same type and color are not needed in a kit. I will also make sure that there are enough parts in the environment in case a team decides to use both arms to grasp two instances of the same part at the same time.

stevegray954, 2020-05-10:
Just was banging my head against something silly with this. I use the left arm to place a part and it drops.  I pick up the dropped part with the right arm and it immediately drops - but now the space the part would teleport to is occupied by the right arm, so the part shoots off into space.

avidacs, 2020-05-11:
Cool! I would suggest to call it instant delivery and award 2 extra points for each. Unless you hit any of the MOs with the shot - in that case your total score is 0.

zeidk, 2020-05-11:
Ok, I will see if Pavel can work on this this week. Trying to use the other arm to regrasp the part will trigger another drop. Regrasping the dropped part with the left arm should not drop the part again based on the plugin behavior. The inconvenient is that it restricts you to use the same arm for both placing and regrasping after a drop.

avidacs, 2020-05-11:
Umh... I think you shouldn't touch the code after submission. Everything was tested on the current docker version...

stevegray954, 2020-05-11:
Yeah, I got it working with the current version by forcing it to use the same arm to regrasp after a drop.  All testing has been against the current docker version, so please don't change it.

zeidk, 2020-05-11:
We are not planning to release any other docker images. We will try to fix it at some point but it will not be pushed until the finals end.

zeidk, 2021-05-02:
Fixed! The plugin started for both robots and both robots had it's own version of parts to drop.

closed 2021-05-02 by zeidk