usnistgov/ARIAC #55: What machines will you be running this on?

stevegray954, 2020-04-24:
I'm noticing very different behavior running inside vs outside of docker -- it seems to be related to simulation speed.  

Out of curiosity, can you please share the specs of the machines you'll be running this on?

zeidk, 2020-04-24:
@stevegray954 Hi Steve, I have attached my machine specs.
Did you get better results inside or outside docker?

[image]

stevegray954, 2020-04-24:
@zeidk I was getting better results outside of Docker.  The issue was with the accuracy of placement of parts being sufficient outside Docker, but inaccurate within (about ~3cm more error in part placement inside the container).  Not sure if its a timing thing (synchronizing the arm and gantry motions) or a controller thing (gains on the gantry -- it does seem to be a bit 'floaty' outside Docker too).
I'm running this on an i7-8850H (6 core, 12 thread) with 32GB RAM and

stevegray954, 2020-04-24:
@zeidk Possibly a silly question, but I cannot find the program you used to generate that hardware details image and the `(unknown)` under graphics has me wondering.  You are running the nvidia drivers, right?

zeidk, 2020-04-25:
@stevegray954 The evaluation will be done solely with docker. We can try running your system inside docker on the same yaml file and see what differences we get from 2 different machines using the same docker image. If you don't mind, **email** me your results and the name of the config file you used and I will let you know the results I get on my side with the same config file. 

It would be even better if you gift me an i7-8850H, you know, just for testing :)

Thanks

stevegray954, 2020-04-25:
@zeidk Thanks Zeid.  I just emailed you.

closed 2021-02-06 by stevegray954