usnistgov/ARIAC #57: pose score for parts with wrong color?

dan9thsense, 2020-05-05:
I am seeing pose scores of 0 for parts that are the wrong color but the correct pose.  For example, deliver red gear parts instead of blue gear parts in order 1 of qual_a_3.  The order calls for:

```
      part_0:
        type: order_part2
        pose:
          xyz: [0.1, -0.2, 0]
          rpy: [0, 0, 0]
      part_1:
        type: order_part2
        pose:
          xyz: [-0.1, -0.2, 0]
          rpy: [0, 0, 0]
```
and here is the output from `rostopic echo /ariac/trays`

```
destination_id: "agv2::kit_tray_2::kit_tray_2::tray"
products: 
  - 
    type: "gear_part_red"
    is_faulty: False
    pose: 
      position: 
        x: -0.102597193884
        y: -0.205921971299
        z: 0.0148429286886
      orientation: 
        x: 0.00778541131923
        y: -0.00481658327204
        z: 0.00435378759387
        w: 0.999948614895
  - 
    type: "gear_part_red"
    is_faulty: False
    pose: 
      position: 
        x: 0.0998754446981
        y: -0.198373526491
        z: 0.0150794675094
      orientation: 
        x: 0.0110030839075
        y: -0.00520408240934
        z: -0.00154785025691
        w: 0.999924724082
```
but the score gives 0 points for pose.

zeidk, 2020-05-06:
Hi Dan, sorry for the late reply.

Pose score affects only parts that are both of correct type and of correct color. If we want to include type and color separately for pose score then we will need to split pose score, ie., 0.5 pts for correct pose and 0.5 pts for correct color.

With the time frame and  things that need to get fixed before the finals I don't think I will be able to implement this any time soon.

dan9thsense, 2020-05-06:
Sounds good, just wanted clarification.

closed 2020-05-06 by dan9thsense