usnistgov/ARIAC #58: ARIAC2021 sample environment takes a long time to load

glpuga, 2021-02-21:
Trying to test the sample environment for ARIAC 2021 using 

> roslaunch nist_gear sample_environment

Takes several minutes of inactivity to load, even when accounting for the time it takes to fetch the models used by the simulation.

The problem seems to be that the configuration file for the environment in `sample_test.yaml` contains two models that are not available to load the current time, `piston_rod_part_blue` and `gear_part_red`.

[image]

**This also impacts testing moveit on the environment**, because the joint controllers timeout while waiting for a service because of the extra time, crashing and leaving the simulation unable to be controlled using rviz/moveit once it finally loads.

Commenting out the section here https://github.com/usnistgov/ARIAC/blob/master/nist_gear/config/sample-test.yaml#L116 allows the simulation to load much faster and allows to test moveit on the environment.

zeidk, 2021-02-22:
A big update will be pushed tomorrow that should fix all the current issues.

glpuga, 2021-02-28:
This is fixed in the latest release, so I'm closing this issue.

closed 2021-02-28 by glpuga