usnistgov/ARIAC #59: seg fault

denbar-robotics, 2021-02-24:
Setup according to the instructions, but seg faults when trying to run
`roslaunch nist_gear sample_environment.launch`

The gazebo screen pops up momentarily and then crashes.  The initial error is:
`[Err] [REST.cc:205] Error in REST request`

Looks like it happens when gazebo gets unpaused:
`gazebo_unpauser.py", line 33`

Here is the error stream:
```

[INFO] [1614202877.670063, 0.000000]: Loading model XML from ros parameter robot_description
[INFO] [1614202877.678518, 0.000000]: Waiting for service /gazebo/spawn_urdf_model
[Err] [REST.cc:205] Error in REST request

libcurl: (51) SSL: no alternative certificate subject name matches target host name 'api.ignitionfuel.org'
[ INFO] [1614202879.546309021]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[INFO] [1614202879.599366, 0.000000]: Calling service /gazebo/spawn_urdf_model
[ WARN] [1614202879.631251693]: spawnURDFModel service called
[ WARN] [1614202879.631973562]: spawnSDFModel service called
[ INFO] [1614202879.636066130]: Physics dynamic reconfigure ready.
[INFO] [1614202879.753070, 0.000000]: Calling service /gazebo/spawn_urdf_model
[INFO] [1614202879.788453, 0.000000]: Calling service /gazebo/spawn_urdf_model
[ WARN] [1614202880.727154377]: spawnURDFModel service called
[INFO] [1614202880.727325, 0.000000]: Spawn status: SpawnModel: Successfully spawned entity
[ WARN] [1614202880.727953991]: spawnSDFModel service called
[spawn_gazebo_sensors-5] process has finished cleanly
log file: /home/dbarry/.ros/log/0485617c-76e9-11eb-8336-181deaec5b6c/spawn_gazebo_sensors-5*.log
[ WARN] [1614202881.004908991]: spawnURDFModel service called
[INFO] [1614202881.005148, 0.000000]: Spawn status: SpawnModel: Successfully spawned entity
[ WARN] [1614202881.005499598]: spawnSDFModel service called
[INFO] [1614202881.006428, 0.000000]: Waiting for service /gazebo/set_model_configuration
[INFO] [1614202881.008359, 0.000000]: temporary hack to **fix** the -J joint position option (issue #93), sleeping for 1 second to avoid race condition.
Service call failed: transport error completing service call: unable to receive data from sender, check sender's logs for details
[ERROR] [1614202881.144598, 0.000000]: Spawn service failed. Exiting.
Segmentation fault
================================================================================REQUIRED process [gazebo-2] has died!
process has died [pid 3312, exit code 139, cmd /home/dbarry/ariac_ws/src/ariac-gazebo_ros_pkgs/gazebo_ros/scripts/gzserver -u -e ode /tmp/ariac/ariac.world __name:=gazebo __log:=/home/dbarry/.ros/log/0485617c-76e9-11eb-8336-181deaec5b6c/gazebo-2.log].
log file: /home/dbarry/.ros/log/0485617c-76e9-11eb-8336-181deaec5b6c/gazebo-2*.log
Initiating shutdown!
================================================================================
**[gazebo_unpauser-15] killing on exit
Traceback (most recent call last):
  File "~/ariac_ws/src/ARIAC/nist_gear/script/gazebo_unpauser.py", line 33, in <module>
    time.sleep(3)**
```

zeidk, 2021-02-25:
I cannot reproduce it on my side. I will take a look today.

zeid

denbar-robotics, 2021-02-25:
Made some progress.  I got rid of this error:
[Err] [REST.cc:205] Error in REST request
by using the suggestion from [here](https://answers.gazebosim.org//question/25030/gazebo-error-restcc205-error-in-rest-request/):

Specifically, the ~/.ignition/fuel/config.yaml file still points to the old ignition fuel domain. This error can be solved by replacing url: https://api.ignitionfuel.org by url: https://api.ignitionrobotics.org inside the ~/.ignition/fuel/config.yaml file.

However, I still get the seg fault
```
INFO] [1614220376.720636, 0.000000]: Controller Spawner: Waiting for service /ariac/kitting/controller_manager/unload_controller
[INFO] [1614220376.723087, 0.000000]: Loading controller: kitting_arm_controller
[ WARN] [1614220376.790398446]: spawnURDFModel service called
[INFO] [1614220376.790623, 0.000000]: Spawn status: SpawnModel: Successfully spawned entity
[ WARN] [1614220376.790712535]: spawnSDFModel service called
[INFO] [1614220376.792132, 0.000000]: Waiting for service /gazebo/set_model_configuration
[INFO] [1614220376.793908, 0.000000]: temporary hack to **fix** the -J joint position option (issue #93), sleeping for 1 second to avoid race condition.
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/controller_manager/spawner", line 212, in <module>
    if __name__ == '__main__': main()
  File "/opt/ros/melodic/lib/controller_manager/spawner", line 190, in main
Service call failed: transport error completing service call: unable to receive data from sender, check sender's logs for details
    Traceback (most recent call last):
  File "/opt/ros/melodic/lib/controller_manager/spawner", line 212, in <module>
resp = load_controller(name)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 442, in __call__
    if __name__ == '__main__': main()
  File "/opt/ros/melodic/lib/controller_manager/spawner", line 204, in main
    resp = switch_controller(loaded, [], 2, False, 0.0)
      File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 442, in __call__
return self.call(*args, **kwds)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 532, in call
[ERROR] [1614220376.920385, 0.000000]: Spawn service failed. Exiting.
    return self.call(*args, **kwds)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 532, in call
    raise ServiceException("transport error completing service call: %s"%(str(e)))
rospy.service.ServiceException: transport error completing service call: unable to receive data from sender, check sender's logs for details
    raise ServiceException("transport error completing service call: %s"%(str(e)))
rospy.service.ServiceException: transport error completing service call: unable to receive data from sender, check sender's logs for details
[INFO] [1614220376.920708, 0.000000]: Shutting down spawner. Stopping and unloading controllers...
[INFO] [1614220376.920807, 0.000000]: Shutting down spawner. Stopping and unloading controllers...
[INFO] [1614220376.922295, 0.000000]: Stopping all controllers...
[INFO] [1614220376.922379, 0.000000]: Stopping all controllers...
Segmentation fault
[WARN] [1614220376.931146, 0.000000]: Controller Spawner error while taking down controllers: unable to connect to service: [Errno 104] Connection reset by peer
[WARN] [1614220376.931214, 0.000000]: Controller Spawner error while taking down controllers: unable to connect to service: [Errno 104] Connection reset by peer
================================================================================REQUIRED process [gazebo-2] has died!
process has died [pid 26291, exit code 139, cmd /home/dbarry/ariac_ws/src/ariac-gazebo_ros_pkgs/gazebo_ros/scripts/gzserver -u -e ode /tmp/ariac/ariac.world __name:=gazebo __log:=~/.ros/log/c0d4a540-7711-11eb-8336-181deaec5b6c/gazebo-2.log].
log file: /home/dbarry/.ros/log/c0d4a540-7711-11eb-8336-181deaec5b6c/gazebo-2*.log
Initiating shutdown!
================================================================================
```

zeidk, 2021-02-25:
I've been trying to pinpoint the issue but to no avail. I wonder if you are missing a package. I will give it a fresher look tomorrow.

denbar-robotics, 2021-02-25:
Perhaps a path or some other environment variable?  Can you provide examples for .bashrc?
Gazebo opens OK on its own with a blank world.

denbar-robotics, 2021-02-25:
Good idea, that installed a bunch of stuff, but the error persists.  I can see the gazebo world load and populated but just for an instant before it crashes.

I tried uninstalling and then reinstalling gazebo, but that did not fix it either.

This works to bring up a blank world in gazebo 9:

`rosrun gazebo_ros gazebo`

zeidk, 2021-02-25:
> @zeidk You're missing `ros-melodic-gazebo-ros-control*` from the list of stuff to install. The `ros_control` packages without Gazebo in the name have nothing to do with Gazebo simulated robots, only real robots.

Thanks, will add it.

zeid

zeidk, 2021-02-25:
@denbar-robotics Trying to finish something and will look into this issue one more time. Is there any log file you can provide? When the environment shows up and crashes, do you get the same errors you posted earlier?

denbar-robotics, 2021-02-25:
I modified gear.launch.template and am able to get the environment to load.

```
  <group ns="ariac/gantry">
  <!-- <param name="tf_prefix" value="gantry_tf" /> -->
    <include file="$(find nist_gear)/launch/gantry.launch">
        <arg name="gantry_urdf_xacro" value="$(arg gantry_urdf_xacro)"/>
        <arg name="limited" value="$(arg limited)"/>
    </include>
    <include file="$(find gantry_moveit_config)/launch/move_group.launch" if="$(arg load_moveit)">
        <arg name="load_robot_description" value="False"/>
    </include>
        <include file="$(find gantry_moveit_config)/launch/move_group.launch" if="$(arg load_gantry_moveit)">
        <arg name="load_robot_description" value="False"/>
    </include>
  </group>

    <group ns="ariac/kitting">
    <!-- <param name="tf_prefix" value="kitting_tf" /> -->
    <include file="$(find nist_gear)/launch/kitting.launch">
    <arg name="kitting_urdf_xacro" value="$(arg kitting_urdf_xacro)"/>
     <arg name="limited" value="$(arg limited)" />
    </include>
    <include file="$(find kitting_moveit_config)/launch/move_group.launch" if="$(arg load_moveit)">
        <arg name="load_robot_description" value="False"/>
    </include>
        <include file="$(find kitting_moveit_config)/launch/move_group.launch" if="$(arg load_kitting_moveit)">
        <arg name="load_robot_description" value="False"/>
    </include>
  </group>

  @[if options['spawn_extra_models']]@
    <!-- spawn any other models -->
  @[for model_name, model in models_to_spawn.items()]@
  @{
initial_pose_string = \
  ' '.join([val for pair in zip(['-x', '-y', '-z'], model.pose.xyz) for val in pair]) + " " + \
  ' '.join([val for pair in zip(['-R', '-P', '-Y'], model.pose.rpy) for val in pair])
  }@

    <node name="nist_gear_spawn_@(model_name.replace('|', '_'))" pkg="gazebo_ros" type="spawn_model"
      args="
        -file $(find nist_gear)/models/@(model.type)_ariac/model.sdf
        -sdf
        -reference_frame @(model.reference_frame)
        @(initial_pose_string)
        -model @(model_name)" />
  @[end for]@
  @[end if]@

  <node name="gazebo_unpauser" pkg="nist_gear" type="gazebo_unpauser.py" output="screen"/>
```

And get this output:

```

process[startup_ariac-1]: started with pid [17800]
Ensuring scoring log file exists
File already exists: /home/dbarry/.ariac/log/performance.log
process[gazebo-2]: started with pid [17802]
process[gazebo_gui-3]: started with pid [17807]
process[sensors_tf_publisher-4]: started with pid [17812]
process[spawn_gazebo_sensors-5]: started with pid [17813]
process[gazebo_unpauser-6]: started with pid [17815]
[startup_ariac-1] process has finished cleanly
log file: /home/dbarry/.ros/log/600ac99e-7791-11eb-bb1c-181deaec5b6c/startup_ariac-1*.log
No handlers could be found for logger "rosout"
[ INFO] [1614276247.284197239]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1614276247.285593655]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[ INFO] [1614276247.315066021]: Finished loading Gazebo ROS API Plugin.
[INFO] [1614276247.683529, 0.000000]: Loading model XML from ros parameter sensors_description
[INFO] [1614276247.686678, 0.000000]: Waiting for service /gazebo/spawn_urdf_model
[INFO] [1614276249.795621, 0.000000]: Calling service /gazebo/spawn_urdf_model
[ INFO] [1614276249.801036623]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ WARN] [1614276249.839886585]: spawnURDFModel service called
[ WARN] [1614276249.840052346]: spawnSDFModel service called
[INFO] [1614276250.934366, 0.000000]: Spawn status: SpawnModel: Successfully spawned entity
[ INFO] [1614276250.940705327]: Physics dynamic reconfigure ready.
[spawn_gazebo_sensors-5] process has finished cleanly
log file: /home/dbarry/.ros/log/600ac99e-7791-11eb-bb1c-181deaec5b6c/spawn_gazebo_sensors-5*.log
[ WARN] [1614276253.036353350]: unpausePhysics service called
[gazebo_unpauser-6] process has finished cleanly
log file: /home/dbarry/.ros/log/600ac99e-7791-11eb-bb1c-181deaec5b6c/gazebo_unpauser-6*.log
```

but when I add in:

```
  <group ns="ariac/gantry">
  <!-- <param name="tf_prefix" value="gantry_tf" /> -->
    <include file="$(find nist_gear)/launch/gantry.launch">
        <arg name="gantry_urdf_xacro" value="$(arg gantry_urdf_xacro)"/>
        <arg name="limited" value="$(arg limited)"/>
    </include>
    <include file="$(find gantry_moveit_config)/launch/move_group.launch" if="$(arg load_moveit)">
        <arg name="load_robot_description" value="False"/>
    </include>
        <include file="$(find gantry_moveit_config)/launch/move_group.launch" if="$(arg load_gantry_moveit)">
        <arg name="load_robot_description" value="False"/>
    </include>
  </group>
```

it crashes with this output:

```
[ WARN] [1614276559.134433128]: spawnURDFModel service called
[ WARN] [1614276559.134601380]: spawnSDFModel service called
[INFO] [1614276560.289736, 0.000000]: Spawn status: SpawnModel: Successfully spawned entity
[ WARN] [1614276560.290308151]: spawnURDFModel service called
[ WARN] [1614276560.291144400]: spawnSDFModel service called
[ INFO] [1614276560.295646061]: Physics dynamic reconfigure ready.
[spawn_gazebo_sensors-5] process has finished cleanly
log file: /home/dbarry/.ros/log/600ac99e-7791-11eb-bb1c-181deaec5b6c/spawn_gazebo_sensors-5*.log
[INFO] [1614276560.588259, 0.000000]: Spawn status: SpawnModel: Successfully spawned entity
[INFO] [1614276560.589518, 0.000000]: Waiting for service /gazebo/set_model_configuration
[INFO] [1614276560.595706, 0.000000]: temporary hack to **fix** the -J joint position option (issue #93), sleeping for 1 second to avoid race condition.
Segmentation fault
```

denbar-robotics, 2021-02-25:
I have isolated the problem to a node in gantry.launch
This is the call that generates the error.  I don't see any relevant log files:
```
    <!-- Run a python script to the send a service call to gazebo_ros to spawn a URDF robot-->
    <node name="gantry_spawner" pkg="gazebo_ros" type="spawn_model" respawn="false" output="screen"
          args="-urdf -model gantry -param robot_description
	  -J long_rail_2_small_rail_joint -6
    -J small_rail_torso_base_joint 0
	  -J torso_base_torso_main_joint 4.65
    -J gantry_arm_shoulder_pan_joint -0.09
    -J gantry_arm_shoulder_lift_joint -0.53
    -J gantry_arm_elbow_joint 1.36
    -J gantry_arm_wrist_1_joint -0.63
    -J gantry_arm_wrist_2_joint -4.72
    -J gantry_arm_wrist_3_joint 0.83"/>
```

And here are the key messages:

```
[INFO] [1614277128.769871, 0.000000]: temporary hack to **fix** the -J joint position option (issue #93), sleeping for 1 second to avoid race condition.
Service call failed: transport error completing service call: unable to receive data from sender, check sender's logs for details
Service call failed: transport error completing service call: unable to receive data from sender, check sender's logs for details
[ERROR] [1614277128.895853, 0.000000]: Spawn service failed. Exiting.
[ERROR] [1614277128.895873, 0.000000]: Spawn service failed. Exiting.
Segmentation fault
```

zeidk, 2021-02-25:
I will do another push today. Let me know if you still have the same issues after it is done.

denbar-robotics, 2021-02-25:
I was able to get it to work on another machine, the one I used for ARIAC 2020, so I guess it is something with my current machine.  Oddly enough, ARIAC 2020 runs fine on the current machine.

zeidk, 2021-02-26:
The first 3 joints listed in gantry.launch are incorrectly named. By the way, in your Home directory there is a .ariac directory which has a performance.log file. This should give you a summary what is running behind the scenes.
We will be pushing a big update today which will take care of many issues. I am finishing the documentation and we will push the new version.

denbar-robotics, 2021-02-28:
That could have caused it.  I'll wait a while until the code is more stable.

denbar-robotics, 2021-03-02:
Unfortunately, the seg fault remains after the big push.  ARIAC 2020 runs fine on the same machine.  Perhaps it is some sort of race condition, as the comment suggests.   Fortunately, I have a different machine where the sim runs OK.

zeidk, 2021-03-02:
@denbar-robotics Does this happen all the time or once in a while? After the seg fault, what happens if you try to run the simulation a few more times? Do you mind attaching performance.log found in ~/.ariac? Thank you.

denbar-robotics, 2021-03-02:
Every time.  Running the sim a few more times does not change the result.  Here is the log file.
[performance.log](https://github.com/usnistgov/ARIAC/files/6070367/performance.log)

dan9thsense, 2021-03-02:
Here is the performance log from the machine where it is working.
[performance.log](https://github.com/usnistgov/ARIAC/files/6070445/performance.log)

zeidk, 2021-03-04:
Are you running a vpn on the machine where it crashes by any chance?

denbar-robotics, 2021-03-04:
Nope.  And, as I mentioned, the machine runs ARIAC 2020 just fine.  I see that someone else has got the same error now. #64

nmullane, 2021-03-05:
I am also running into this error. I'm running in a VM at the moment, but I am going to see if I still get the error when I install 18.04 on my desktop again.

zeidk, 2021-03-06:
A few things to try to narrow down the issue.

### REST issue

Open ~/.ignition/fuel/config.yaml and let me know what entry you have for the key 'url'.
You probably have the following:
```yaml
servers:
  -
    name: osrf
    url: https://api.ignitionfuel.org
```

Edit this file so the entry looks like what we have below.
- Report back whether or not the REST issue is gone.

```yaml
servers:
  -
    name: osrf
    # url: https://api.ignitionfuel.org
    url: https://api.ignitionrobotics.org
```

### Gazebo Control

Double check the [GEAR dependencies](https://github.com/usnistgov/ARIAC/blob/master/wiki/tutorials/installation.md#install-gear-dependencies).

On top of those commands, try `sudo apt-get install ros-melodic-gazebo-ros-pkgs` to get all the necessary gazebo plugins.

- Report back whether new packages were installed or not after issuing those commands.

### Controller Manager for Melodic

Check the controller manager for Melodic with `dpkg -l | grep melodic-controller-manager`

- Report back with the result.

Here are my results:

```bash
ii  ros-melodic-controller-manager                0.18.3-1bionic.20201207.160633                   amd64        The controller manager.
ii  ros-melodic-controller-manager-msgs           0.18.3-1bionic.20201207.151258                   amd64        Messages and services for the controller manager.
```

The most important parts are that the start numbers (`0.18.3`) of the manager and the manager-msgs match. 

### Version of Gazebo

Which version of Gazebo are you running?

denbar-robotics, 2021-03-06:
Looks pretty much the same as yours.

 dpkg -l | grep melodic-controller-manager
ii  ros-melodic-controller-manager               0.18.3-1bionic.20201207.160633                   amd64        
ii  ros-melodic-controller-manager-dbgsym        0.18.3-1bionic.20201207.160633                   amd64      
ii  ros-melodic-controller-manager-msgs          0.18.3-1bionic.20201207.151258                   amd64       
ii  ros-melodic-controller-manager-tests         0.18.3-1bionic.20201207.161537                   amd64        
ii  ros-melodic-controller-manager-tests-dbgsym  0.18.3-1bionic.20201207.161537                   amd64       

gazebo 9.0.0

stevegray954, 2021-03-06:
@dan9thsense  @zeidk That's our answer!

It works for me, and I'm running Gazebo 9.16.0.
I had forgotten that they switched the ignition/fuel address over during the 9.0 releases.  Installing ROS just installs the base version that came with that ROS release, but because the Gazebo apt repo is different than the ROS one, that package is never updated.

You'll want to follow these steps from http://gazebosim.org/tutorials?tut=install_ubuntu to update Gazebo:
```
sudo sh -c 'echo "deb http://packages.osrfoundation.org/gazebo/ubuntu-stable `lsb_release -cs` main" > /etc/apt/sources.list.d/gazebo-stable.list'
wget https://packages.osrfoundation.org/gazebo.key -O - | sudo apt-key add -
sudo apt update
sudo apt upgrade
```

zeidk, 2021-03-06:
@dan9thsense 
> gazebo 9.0.0

This is your culprit. As Steve said, 9.0 will not work and I would suggest you update to 9.16 as well (I will emphasize this in the install section of the wiki). The other package that needs to be cloned along the ARIAC package (see [here](https://github.com/usnistgov/ARIAC/blob/master/wiki/tutorials/installation.md#download-source-code)) is a custom Gazebo package that ARIAC uses. If you remove `--development-mode` from `sample_environment.launch` the robots will not span in the environment. In summary:
 - Update to Gazebo 9.16.
 - Make sure the gazebo package is in your catkin workspace.

dan9thsense, 2021-03-06:
Upgrading gazebo fixed it.  I already had the custom Gazebo package and thought it was clear in the instructions it is needed. 
 Oh the other hand, note that the installation instructions do not mention a particular version of Gazebo 9.  They just say:
`Follow these instructions to install Gazebo 9.`

Thanks for the hard work in figuring it out.  Sometimes the simplest causes can be hard to find.

zeidk, 2021-03-06:
It is especially much harder to figure out the source of the issues if you cannot reproduce it on your side.

Thanks for the help Steve.

zeidk, 2021-03-09:
Closing this one as it seems to have fixed the original issue.

closed 2021-03-09 by zeidk