usnistgov/ARIAC #61: scoring

dan9thsense, 2021-03-02:
I found the section that describes scoring for parts on the tray, but it does not mention how the scoring works for assembled parts.

denbar-robotics, 2021-03-02:
I realized that the assembly has the same paramters as the tray

zeidk, 2021-03-02:
They have similar scoring metrics but I will update the page to make it clear.

zeidk, 2021-03-02:
While we are making updates to the wiki here is a summary on assembly scoring:

"For the Assembly process, the completion score will be similarly calculated, with 2 points awarded for each part being successfully assembled, with a double all-parts bonus to make a maximum of 4 times the number of parts being inserted into the assembly (so for the ventilator briefcase design, it would be a maximum of 16 points for a fully completed ventilator) (4 x 4 = 16)."

denbar-robotics, 2021-03-02:
Any points for getting a part into the briefcase, but not correctly oriented?

zeidk, 2021-03-03:
> Any points for getting a part into the briefcase, but not correctly oriented?

@anthonydowns Any thoughts here?

We are also working on the assembly plugin to keep parts in place once positioned. Just letting you know since you are currently focusing on assembly.

zeidk, 2021-04-20:
Documentation was updated a while ago regarding this question. Closing this one. Let me know if the documentation on scoring does not answer this question.

closed 2021-04-20 by zeidk