usnistgov/ARIAC #63: tf tree is disconnected

dan9thsense, 2021-03-03:
The logical camera is not connected to the rest of the tf tree.  The breakbeam is connected.
The camera appears correctly in gazebo.  The tf tree from `rosrun tf view_frames` is attached.
[image]

Here is the config file:
sensors:
  break_beam_1:
    type: break_beam
    pose:
      xyz: [1.6, 2.25, 0.91]
      rpy: [0, 0, 'pi']

  logical_camera_0:
    type: logical_camera
    pose:
      xyz: [-2.515033, 2.925223, 1.82]
      rpy: [-1.570796, 1.570796, 0]

zeidk, 2021-03-03:
Hi Dan, please see this merged pull request #31

dan9thsense, 2021-03-03:
I put a camera over some objects on a tray and the topic publishes and the transform from the object to the camera publishes, but I am still unable to get the transform from the camera to the world.  Are the static transforms publishing?  Normally in ROS there should not be disconnected tf trees.

`rosrun tf tf_echo logical_camera_0_frame logical_camera_0_assembly_regulator_red_4_frame
- Translation: [1.065, -0.715, 0.554]
- Rotation: in Quaternion [-0.495, 0.507, 0.498, -0.500]
            in RPY (radian) [1.575, -0.014, -1.582]
            in RPY (degree) [90.246, -0.781, -90.629]

rosrun tf tf_echo logical_camera_0_frame world
Exception thrown:Could not find a connection between 'logical_camera_0_frame' and 'world' because they are not part of the same tree.Tf has two or more unconnected trees.`

dan9thsense, 2021-03-04:
I got it working by publishing the static transforms myself, but, of course, they depend on the user config.

zeidk, 2021-03-04:
Seeing your message now.  Competitors should not have to do extra work to publish the transforms. I'll take a look.

=

zeidk, 2021-03-04:
Dan, can you query the logical camera with a regular 'rostopic echo' ?
If so, you can get the pose of the camera in the world frame without doing any transform. Logical cameras report poses of parts found + pose of the cameras themselves in the world frame. Below is an example of outputs retrieved from a logical camera.

```bash
models: 
  - 
    type: "assembly_regulator_red"
    pose: 
      position: 
        x: 1.06173082606
        y: -0.71677474415
        z: 0.555114938791
      orientation: 
        x: 0.497786462927
        y: -0.503666001055
        z: -0.504150627333
        w: 0.494329183508
pose: 
  position: 
    x: -2.515033
    y: 2.925223
    z: 1.82
  orientation: 
    x: -0.5
    y: 0.5
    z: 0.499999836603
    w: 0.500000163397
---
```

The last entry in the output above (`pose`) is the pose of the logical camera itself in the world frame.

stevegray954, 2021-03-04:
Are the logical camera transforms no longer on /tf_static ?

stevegray954, 2021-03-04:
Looking closer, it looks like the logical cameras are not showing up in the  `sensors_description` rosparam anymore, but all other user-specified and other sensors are.  This would explain why they don't get a static tf published by this guy

```
  <node pkg="robot_state_publisher" type="robot_state_publisher" name="sensors_tf_publisher">
    <remap from="robot_description" to="sensors_description" />
    <param name="publish_frequency" type="double" value="50.0" />
    <param name="tf_prefix" type="string" value="" />
  </node>
```

zeidk, 2021-03-04:
> `&#x002D;&#x002D;`

From OSRF since 2017. ASCII for -
`&#x002D;&#x002D;inorder` would be `--inorder`

stevegray954, 2021-03-04:
Ah.  Anyway, looking at the generated `sensors_description` param, it's missing all the logical cameras, but has the breakbeams, laser profilers, quality control sensors, etc.  I'm trying to figure out why the logical cameras are missing now.

zeidk, 2021-03-04:
There are a few places where logical cameras are mentioned in the whole repository. 

https://github.com/usnistgov/ARIAC/blob/f1dbe0c3be4798d5eeb44d4544796f3f5884cdd3/nist_gear/launch/gear.urdf.xacro.template#L8

https://github.com/usnistgov/ARIAC/blob/f1dbe0c3be4798d5eeb44d4544796f3f5884cdd3/nist_gear/worlds/ariac.world.template#L317

https://github.com/usnistgov/ARIAC/blob/f1dbe0c3be4798d5eeb44d4544796f3f5884cdd3/nist_gear/worlds/snippet/logical_camera.sdf.template#L1

@Pavel-P Any recent changes that may have affected this?

stevegray954, 2021-03-04:
so far it all seems fine through `prepare_template_data` in `gear.py`.  Continuing to track.

stevegray954, 2021-03-04:
Boom.

```  <!-- sensor links and fixed joints -->
@[for name, sensor in sensors.items()]@
@[    if not name.startswith('logical_camera')  # logical cameras publish their own TF pose dynamically]@

```
someone commented it out for some reason in `gear.urdf.xacro.template`

stevegray954, 2021-03-04:
See PR #65

Works in my testing

stevegray954, 2021-03-05:
@zeidk please close assuming #65 solved it

dan9thsense, 2021-03-06:
Yes, that fixed it.  Thanks Steve!

closed 2021-03-06 by zeidk