usnistgov/ARIAC #64: Error/Segmentattion fault spawning the ganty

GerardHarkemaAvans, 2021-03-04:
When starting the sample_environment.launch gazebo crashes. After some searching i find out that spawning the gantry cause this crash. It is in the file "gantry.launch". Commenting out the following lines stops the crashes but the gantry is not loaded into Gazebo.

    
    <node name="gantry_spawner" pkg="gazebo_ros" type="spawn_model" respawn="false" output="screen"
          args="-urdf -model gantry -param robot_description
	  -J small_long_joint -1.45
    -J torso_rail_joint 0
	  -J torso_base_main_joint 0
    -J torso_main_torso_tray_joint 0
    -J gantry_arm_shoulder_pan_joint 0.00
    -J gantry_arm_shoulder_lift_joint -1.13
    -J gantry_arm_elbow_joint 1.88
    -J gantry_arm_wrist_1_joint -0.72
    -J gantry_arm_wrist_2_joint 1.55
    -J gantry_arm_wrist_3_joint 0.83"/>

zeidk, 2021-03-04:
Hello, can you please try removing only `-J torso_main_torso_tray_joint 0` and see if it loads?

GerardHarkemaAvans, 2021-03-04:
Sorry, tried your suggestion, loading still fails. Removed all joint all and it  still won´t load.

Hope this dump will help you.
... logging to /home/gerard/.ros/log/dbe7c5ce-7d33-11eb-87e8-b8aeedf4e998/roslaunch-gerard-ubuntu-4142.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://gerard-ubuntu:41485/

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.10

NODES
  /
    ariac_sim (nist_gear/gear.py)

auto-starting new master
process[master]: started with pid [4152]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to dbe7c5ce-7d33-11eb-87e8-b8aeedf4e998
process[rosout-1]: started with pid [4163]
started core service [/rosout]
process[ariac_sim-2]: started with pid [4170]
writing file /tmp/ariac/gear.urdf.xacro
writing file /tmp/ariac/ariac.world
writing file /tmp/ariac/gear.launch
writing file /tmp/ariac/gantry.urdf.xacro
writing file /tmp/ariac/kitting.urdf.xacro
Running command: roslaunch /tmp/ariac/gear.launch world_path:=/tmp/ariac/ariac.world gear_urdf_xacro:=/tmp/ariac/gear.urdf.xacro gantry_urdf_xacro:=/tmp/ariac/gantry.urdf.xacro kitting_urdf_xacro:=/tmp/ariac/kitting.urdf.xacro
... logging to /home/gerard/.ros/log/dbe7c5ce-7d33-11eb-87e8-b8aeedf4e998/roslaunch-gerard-ubuntu-4173.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
xacro: in-order processing became default in ROS Melodic. You can drop the option.
xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://gerard-ubuntu:33589/

SUMMARY
========

PARAMETERS
 * /ariac/gantry/gantry_arm_controller/action_monitor_rate: 10
 * /ariac/gantry/gantry_arm_controller/constraints/gantry_arm_elbow_joint/goal: 0.1
 * /ariac/gantry/gantry_arm_controller/constraints/gantry_arm_elbow_joint/trajectory: 0.1
 * /ariac/gantry/gantry_arm_controller/constraints/gantry_arm_shoulder_lift_joint/goal: 0.1
 * /ariac/gantry/gantry_arm_controller/constraints/gantry_arm_shoulder_lift_joint/trajectory: 0.1
 * /ariac/gantry/gantry_arm_controller/constraints/gantry_arm_shoulder_pan_joint/goal: 0.1
 * /ariac/gantry/gantry_arm_controller/constraints/gantry_arm_shoulder_pan_joint/trajectory: 0.1
 * /ariac/gantry/gantry_arm_controller/constraints/gantry_arm_wrist_1_joint/goal: 0.1
 * /ariac/gantry/gantry_arm_controller/constraints/gantry_arm_wrist_1_joint/trajectory: 0.1
 * /ariac/gantry/gantry_arm_controller/constraints/gantry_arm_wrist_2_joint/goal: 0.1
 * /ariac/gantry/gantry_arm_controller/constraints/gantry_arm_wrist_2_joint/trajectory: 0.1
 * /ariac/gantry/gantry_arm_controller/constraints/gantry_arm_wrist_3_joint/goal: 0.1
 * /ariac/gantry/gantry_arm_controller/constraints/gantry_arm_wrist_3_joint/trajectory: 0.1
 * /ariac/gantry/gantry_arm_controller/constraints/goal_time: 0.6
 * /ariac/gantry/gantry_arm_controller/constraints/stopped_velocity_tolerance: 0.05
 * /ariac/gantry/gantry_arm_controller/gains/gantry_arm_elbow_joint/d: 150
 * /ariac/gantry/gantry_arm_controller/gains/gantry_arm_elbow_joint/i: 1
 * /ariac/gantry/gantry_arm_controller/gains/gantry_arm_elbow_joint/i_clamp: 25
 * /ariac/gantry/gantry_arm_controller/gains/gantry_arm_elbow_joint/p: 50000
 * /ariac/gantry/gantry_arm_controller/gains/gantry_arm_shoulder_lift_joint/d: 150
 * /ariac/gantry/gantry_arm_controller/gains/gantry_arm_shoulder_lift_joint/i: 10
[... 244 lines cut ...]
[ariac/kitting/kitting_spawner-12] killing on exit
[ariac/gantry/gantry_controller_spawner-9] killing on exit
[ariac/gantry/robot_state_publisher-7] killing on exit
[ariac/gantry/joint_state_controller_spawner-6] killing on exit
[sensors_tf_publisher-4] killing on exit
[ariac/kitting/joint_state_controller_spawner-10] killing on exit
[gazebo_gui-3] killing on exit
[gazebo-2] killing on exit
[WARN] [1614894735.478588, 0.000000]: Controller Spawner couldn't find the expected controller_manager ROS interface.
[WARN] [1614894735.527391, 0.000000]: Controller Spawner couldn't find the expected controller_manager ROS interface.
[gazebo_gui-3] escalating to SIGTERM
shutting down processing monitor...
... shutting down processing monitor complete
done
================================================================================REQUIRED process [ariac_sim-2] has died!
process has finished cleanly
log file: /home/gerard/.ros/log/dbe7c5ce-7d33-11eb-87e8-b8aeedf4e998/ariac_sim-2*.log
Initiating shutdown!
================================================================================
[ariac_sim-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

zeidk, 2021-03-05:
I managed to reproduce the error on another computer. We are working on a fix.

dan9thsense, 2021-03-06:
I made that change (see #59) and it helped with the REST error, but I still got a seg fault

zeidk, 2021-03-06:
Since your issue is similar to the one described in #59 I will comment in the other ticket and not here. Please refer to my comments in the other ticket.

stevegray954, 2021-03-06:
@GerardHarkemaAvans Please try updating your Gazebo version, as I posted about here:
https://github.com/usnistgov/ARIAC/issues/59?_pjax=%23js-repo-pjax-container#issuecomment-791934930

GerardHarkemaAvans, 2021-03-07:
Yes!!!
This works. Thanks a lot.
Kind regards,

Gerard

closed 2021-03-07 by zeidk