usnistgov/ARIAC #66: Will new order event only be triggered in the kitting task?

barrygxwan, 2021-03-07:
if the order_0 only has one shipment with assembly task, is it passable to trigger the New order event?

zeidk, 2021-03-07:
New Order occurs only after you deliver a kit (i.e.,  order_0 has at least one kitting task).

barrygxwan, 2021-03-08:
thank you very much，zeidk. I have a another question: Will the collision between the robot and the sensor affect the score factor COL?

zeidk, 2021-03-08:
No, we have removed all collisions on the sensors. You do not have to worry about colliding with any sensors. We have updated the scoring section and we will push a new update this week (hopefully by Wednesday). COL will set to 0 if the robots collide with each other. COL will also be set to 0 if the gantry arm collides with the gantry torso.

barrygxwan, 2021-03-08:
OK，thank you. and I found  the NOTE for Part Re-orientation on https://github.com/usnistgov/ARIAC/blob/master/wiki/documentation/agility_challenges.md  mentioned that See documentation on [frame specifications](https://github.com/usnistgov/ARIAC/blob/master/wiki/documentation/frame_specifications.md#flipped-products) for details on re-orienting products.
the link page says :Teams are not permitted to directly grasp this product from the side when a product flip is required. this rule is for 2020, and is it also applies to this year?

barrygxwan, 2021-03-08:
> New Order can occur only after you deliver a kit (i.e., order_0 has at least one kitting task).
> 
> If order_0 has only assembly then you will not have New Order.
> 
> If order_0 has both assembly and kitting then New Order can occur after the kitting shipment is delivered.

hi, zeidk. I feel a little confused about “order_0 has at least one kitting task” and “If order_0 has only assembly then you will not have New Order.”

zeidk, 2021-03-09:
An order can have up to 2 shipments: 
 - 1.  kitting only (e.g., sample_kitting.yaml), 
 - 2. assembly only (e.g., sample_assembly.yaml), 
 - 3. kitting and assembly (e.g., sample_kitting_assembly.yaml). 

 If the order has only assembly (ii) then it is certain that New Order will not happen. New Order can happen when order_0 has a kitting task (i or iii).

I hope this makes sense.

barrygxwan, 2021-03-09:
thank you for your reply! I get it !

closed 2021-03-09 by barrygxwan