usnistgov/ARIAC #68: Teams are not permitted to directly grasp this product from the side when a product flip is required？

barrygxwan, 2021-03-09:
Hi，
I found the NOTE for Part Re-orientation on https://github.com/usnistgov/ARIAC/blob/master/wiki/documentation/agility_challenges.md mentioned that See documentation on frame specifications for details on re-orienting products.
the link page says :Teams are not permitted to directly grasp this product from the side when a product flip is required. this rule is for 2020, and is it also applies to this year?

zeidk, 2021-03-09:
Hello, the pump can be grasped from any side if orientation is needed. This statement was for ARIAC 2020 and has been removed in the newest version of the wiki. Thank you.

zeidk, 2021-03-15:
Closing this now since the documentation has been updated. Thank you.

closed 2021-03-15 by zeidk