usnistgov/ARIAC #69: can assembly products be gathered from bins?

denbar-robotics, 2021-03-09:
Or can they only come from AGVs?  If it is only from AGVs, can the kitting arm put products on an AGV that are not part of an order, so that the assembly arm can take them for assembly?

zeidk, 2021-03-09:
When assembly is in order_0, the trial will start with an AGV (or two AGVs) already located at the station where assembly needs to be done. You will not need to build a kit and then deliver the kit to the station. This situation is described in sample_assembly.yaml. In this file, you have two AGVs both located at assembly station 1 (`as1`) when the trial starts. Each AGV has a different part and both parts are needed for the assembly in order_0.

In the scenario  where only kitting is required for order_0: Once the kit is built and delivered to the correct station using the correct AGV you will receive a new order (order_1) asking you to do assembly at the station where the AGV was delivered. If you use the wrong AGV or the wrong station when delivering the AGV this new order will not be triggered. Moreover, this new order's priority is 1, since it's not an agility challenge, It is just another way to announce a new order. Take a look at sample_new_order.yaml which describes this situation.

dan9thsense, 2021-03-09:
Thanks for the clarification.  What if the kitting order was not completely filled, but there are parts still in the bins.  Can the assembly robot use those parts to complete the assembly or do all the parts have to come from the assigned agv?

Are there cases where kitting and assembly are happening simultaneously?  For example, the kitting arm could be supplying agv1 and there could be an assembly order from agv4. Or perhaps the kitting order was completed and the assembly order started and then another kitting order (for a different agv) is received.

zeidk, 2021-03-09:
The use of AGVs for assembly is not enforced but we wanted to make things easier for you. Quality control sensors on AGVs are very helpful. If a delivered AGV is missing some parts you can definitely task the gantry to pick up those missing parts. You need to ensure these parts are not faulty. 
 - Parts on AGVs have zero chance to be faulty if the AGVs are already located at stations when the trial starts (I will take care of them not being faulty).
 - If you are required to do kitting first (order_0) and then assembly is announced in New Order (order_1) you will have a  very low chance to use faulty parts for assembly since you built the kit yourself and you probably took care of faulty parts.

Yes, order_0 can consist of both kitting and assembly. order_0 can also be interrupted by a higher-priority order (wanted and unwanted products). It can also happen that New Order is triggered after delivering the AGV from order_0. In the latter case you will have to do the original assembly (order_0) and another assembly (order_1). We will try to not go too crazy with these kind of scenarios but they can happen.

zeidk, 2021-03-14:
Closing this one. The documentation has been updated based on this question. Thank you.

closed 2021-03-14 by zeidk