usnistgov/ARIAC #70: what‘s the complete [False] item mean?

barrygxwan, 2021-03-09:
[ INFO] [1615274667.080190143, 5.769000000]: [agv2-> as2] SUCCESS: AGV successfully triggered.
[ INFO] [1615274667.084168258, 5.770000000]: All kitting shipments submitted for order: order_0
[ INFO] [1615274667.090118015, 5.771000000]: End of trial. Final score: 10
Score breakdown:
[game_score]
...total game score: [10]
...total process time: [0.741]
...arms collision?: [0]
   [order score]
   ...order ID [order_0]
   ...total order score: [10]
   ...completion score: [10]
   ...time taken: [0.739]
   ...priority: [1]
      [kitting score]
      ...shipment type: [order_0_kitting_shipment_0]
      ...completion score: [10]
      ...complete: [false]         @ this item
      ...submitted: [true]
      ...used correct agv: [true]
      ...sent to correct station: [true]
      ...product type presence score: [3]
      ...product color presence score: [3]
      ...product pose score: [3]
      ...all products bonus: [1]

[ INFO] [1615274667.090620806, 5.771000000]: [kitting]:"order_id","completion_score","priority","time","is_complete","shipments_requested","shipments_submitted","shipments_completed","shipments_wrong_agv", "shipments_wrong_station"
"order_0",10,1,0.739,1,1,1,0,0,0
[assembly]: No assembly shipment was requested in this order

zeidk, 2021-03-09:
This means the shipment does not have all the parts as specified in this order. We fixed some sections in this output in the latest build of the software, which we are planning to push on Wednesday.

zeidk, 2021-03-09:
Yes, the display is not correct and fixed it, no worries.

closed 2021-03-11 by barrygxwan