usnistgov/ARIAC #71: Problem running (example) simulation

AngeloFerrando, 2021-03-10:
Hi,
I have tried to run the simulation (following tutorial's steps), but I get the following error.
Did I forget to set or install anything?
Thank you in advance.

$ roslaunch nist_gear sample_environment.launch
...
[ INFO] [1615381066.337358345]: Failed to call service get_planning_scene, have you launched move_group or called psm.providePlanningSceneService()?
[ INFO] [1615381067.164898772]: Failed to call service get_planning_scene, have you launched move_group or called psm.providePlanningSceneService()?
[WARN] [1615381088.827516, 0.000000]: Controller Spawner couldn't find the expected controller_manager ROS interface.
[WARN] [1615381088.905038, 0.000000]: Controller Spawner couldn't find the expected controller_manager ROS interface.
[ariac/gantry/joint_state_controller_spawner-6] process has finished cleanly
log file: /home/name/.ros/log/2f28d6f4-81a0-11eb-9bad-309c231893df/ariac-gantry-joint_state_controller_spawner-6*.log
[ariac/gantry/gantry_controller_spawner-9] process has finished cleanly
log file: /home/name/.ros/log/2f28d6f4-81a0-11eb-9bad-309c231893df/ariac-gantry-gantry_controller_spawner-9*.log
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/controller_manager/spawner", line 212, in <module>
Service call failed: transport error completing service call: unable to receive data from sender, check sender's logs for details
    if __name__ == '__main__': main()
  File "/opt/ros/melodic/lib/controller_manager/spawner", line 204, in main
    resp = switch_controller(loaded, [], 2, False, 0.0)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 442, in __call__
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/controller_manager/spawner", line 212, in <module>
    return self.call(*args, **kwds)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 532, in call
    if __name__ == '__main__': main()
  File "/opt/ros/melodic/lib/controller_manager/spawner", line 204, in main
    resp = switch_controller(loaded, [], 2, False, 0.0)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 442, in __call__
    raise ServiceException("transport error completing service call: %s"%(str(e)))
rospy.service.ServiceException: transport error completing service call: unable to receive data from sender, check sender's logs for details
[ERROR] [1615381094.576557, 0.000000]: Spawn service failed. Exiting.
    return self.call(*args, **kwds)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 532, in call
    raise ServiceException("transport error completing service call: %s"%(str(e)))
rospy.service.ServiceException: transport error completing service call: unable to receive data from sender, check sender's logs for details
[INFO] [1615381094.576876, 0.000000]: Shutting down spawner. Stopping and unloading controllers...
[INFO] [1615381094.577071, 0.000000]: Shutting down spawner. Stopping and unloading controllers...
[INFO] [1615381094.578706, 0.000000]: Stopping all controllers...
[INFO] [1615381094.578941, 0.000000]: Stopping all controllers...
Service call failed: unable to connect to service: [Errno 104] Connection reset by peer
[ERROR] [1615381094.586497, 0.000000]: SetModelConfiguration service failed. Exiting.
[WARN] [1615381094.586858, 0.000000]: Controller Spawner error while taking down controllers: unable to connect to service: [Errno 104] Connection reset by peer
Service call failed: unable to connect to service: [Errno 104] Connection reset by peer
[ERROR] [1615381094.587529, 0.000000]: SetModelConfiguration service failed. Exiting.
[WARN] [1615381094.587947, 0.000000]: Controller Spawner error while taking down controllers: unable to connect to service: [Errno 104] Connection reset by peer
Segmentation fault (core dumped)
================================================================================REQUIRED process [gazebo-2] has died!
process has died [pid 15281, exit code 139, cmd /media/name/WorkData/git/ariac_ws/src/ariac-gazebo_ros_pkgs/gazebo_ros/scripts/gzserver -u -e ode /tmp/ariac/ariac.world __name:=gazebo __log:=/home/name/.ros/log/2f28d6f4-81a0-11eb-9bad-309c231893df/gazebo-2.log].
log file: /home/name/.ros/log/2f28d6f4-81a0-11eb-9bad-309c231893df/gazebo-2*.log
Initiating shutdown!
================================================================================
[ariac/kitting/kitting_controller_spawner-14] killing on exit
[gazebo_unpauser-15] killing on exit
[ariac/kitting/robot_state_publisher-11] killing on exit
[ariac/kitting/fake_joint_calibration-13] killing on exit
[ariac/gantry/gantry_spawner-8] killing on exit
[ariac/kitting/joint_state_controller_spawner-10] killing on exit
[ariac/kitting/kitting_spawner-12] killing on exit
[spawn_gazebo_sensors-5] killing on exit
[gazebo_gui-3] killing on exit
[sensors_tf_publisher-4] killing on exit
[ariac/gantry/robot_state_publisher-7] killing on exit
[gazebo-2] killing on exit
Segmentation fault (core dumped)
shutting down processing monitor...
... shutting down processing monitor complete
done
================================================================================REQUIRED process [ariac_sim-2] has died!
process has finished cleanly
log file: /home/name/.ros/log/2f28d6f4-81a0-11eb-9bad-309c231893df/ariac_sim-2*.log
Initiating shutdown!
================================================================================
[ariac/kitting/rviz_name_GS63VR_7RF_14447_3843900788838012672-4] killing on exit
[ariac/gantry/rviz_name_GS63VR_7RF_14447_9090571566890448477-3] killing on exit
[ariac_sim-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

johnmichaloski, 2021-03-10:
Are you using gazebo 9.16?I had similar problems with 18.04 base install of gazebo 9. I think the wiki has instructions to upgrade (its tricky).

AngeloFerrando, 2021-03-10:
Yes, I am using gazebo 9. So, by upgrading it to a newer version solved the problem?
Thanks.

johnmichaloski, 2021-03-10:
You will probably get this error after upgrading:
nala@lightning:~/ros/ARIAC/src$ gazebo
gazebo: symbol lookup error: 
/usr/lib/x86_64-linux-gnu/libgazebo_common.so.9: undefined symbol: 
_ZN8ignition10fuel_tools12ClientConfig12SetUserAgentERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE

Its a dependent ignition version problem:
FIX:  sudo apt upgrade libignition-math2 --fix-missing

johnmichaloski, 2021-03-10:
Here are helpful web sites:
https://answers.gazebosim.org/question/24645/how-to-update-gazebo-900-to-latest-after-ros-melodic-full-desktop-install/

and

http://gazebosim.org/tutorials?cat=install&tut=install_ubuntu&ver=9.0

but I had a lot of apt-get  update/upgrade issues in general. So good luck.

AngeloFerrando, 2021-03-10:
Thank you very much

zeidk, 2021-03-10:
Hi Angelo, can you take a look at #59 and see if the problem is similar to yours?

AngeloFerrando, 2021-03-10:
Yes, the problem is similar and the solution seems to work for me too. After upgrading gazebo to 9.16 it seems to work.
Thank you very much!

closed 2021-03-10 by AngeloFerrando