usnistgov/ARIAC #73: is it possible that there are exist some parts on agv , and this agv is the target agv that order_0 requires for kiiting shipment

barrygxwan, 2021-03-11:
hi, I have some questions:
Q1: is it possible that there are exist some parts on agv , and this agv is the target agv that order_0 requires for kiiting shipment?
For example : if order_0 have a kitting shipment and the agv_id is agv1 , and there are some parts on agv1 when start, we need to remove these parts firstly, and then do kitting.  is this possible? 
Q2: I notice that there are two  orientation types of sensor: rpy=[**0**,0,x] and rpy =[**pi/2**,0,x]. So during the competition, Will these two situations happen?
Q3：The drop region beyond agv seems have a problem, when close to the drop region, part always drop on the ground.
Q4:  The most complicated situation for orders : Order_0, Oder_0_update, Order_1, Order_1_update.  am I right ?

zeidk, 2021-03-11:
A1: If the AGV is a target AGV then it will be empty and not loaded with some parts.
A2: I am not sure what you are referring to. Can you point to a file?
A3: We will try to fix this as soon as possible.
A4: This is correct.

barrygxwan, 2021-03-12:
thank you for your reply.
Q2 : file:[https://github.com/usnistgov/ARIAC/blob/master/nist_gear/config/trial_config/sample_new_order.yaml](url) 

models_over_bins 
bin5:models:
       **assembly_sensor_blue:**
         xyz_start: [0.2, 0.2, 0.0]
         xyz_end: [0.4, 0.4, 0.0]
         **rpy: ['pi/2', 0, 'pi/4'**]
         num_models_x: 2
         num_models_y: 1
      file :[https://github.com/usnistgov/ARIAC/blob/master/nist_gear/config/trial_config/sample_faulty_gripper.yaml](url) 
models_over_bins:
   bin5:
     models:
       **assembly_sensor_blue:**
         xyz_start: [0.1, 0.2, 0.0]
         xyz_end: [0.4, 0.5, 0.0]
         **rpy: [0, 0, 'pi/4']**
         num_models_x: 2
         num_models_y: 2

zeidk, 2021-03-13:
We have fixed the roll for all parts. They are all 0 now except when a part needs to be flipped.

zeidk, 2021-03-14:
Please take a look at the new updates and let me know if there are further questions/issues.

closed 2021-03-14 by zeidk