usnistgov/ARIAC #76: there are 2 probloms and 1 question for submitting a task.

dongqingwei, 2021-03-13:
hi, we found 2 probloms:
Problom1: After kitting task finish, if we submit directly, we found the agv is static, and the parts in the agv tray are jumping.
Problom2: After the AGV arrived in the assembly station, we can’t submit the task.
Q: After the last two probloms are solved, will the parts disappear after submission?  Can these parts be used for assembly?

zeidk, 2021-03-13:
Thank you, we have also fixed this. The new update will be pushed today. I can update the 2021 branch and merge it later into master so you can still do your work.

No, the parts will not disappear once you submit an AGV. There are situations where parts on delivered AGVs are needed for assembly.

dongqingwei, 2021-03-14:
Well, thank you very much.

zeidk, 2021-03-14:
Try the new version of the software and let me know if there are other issues or questions. Thank you.

closed 2021-03-14 by zeidk