usnistgov/ARIAC #78: new release breaks a lot of code

dan9thsense, 2021-03-14:
The changes to logical camera coverage in the new release break a lot of existing code.  Teams that went to a lot of effort to use the previous coverage dimensions now find that work useless.  While the larger coverage makes the layout easier and less costly, it penalizes teams that spent a lot of effort to make the previous coverage work.

Similarly, now charging a cost for the gantry camera throws out a lot of work that was done in cost analysis.  That has to be redone and, if a different decision is reached based on the larger logical camera coverage plus the cost of the gantry camera, then the code written to utilize the gantry camera becomes a large waste of time.

Please, please, do not make fundamental changes to the rules and features after teams have been working based on the original rules and features.  These changes did not fix any bugs, so they were not necessary, and it is very frustrating to see a lot of work go out the window for what seem to be arbitrary changes.

zeidk, 2021-03-14:
Thanks for the feedback Dan. We usually revert back to the previous version when a customer is not satisfied. Let me talk to the team on Monday first. On your system you can revert the fov of the logical camera back in `worlds/snippet/logical_camera.sdf.template` and modify lines 59--64 with the following:

```xml
          <logical_camera>
            <near>0.2</near>
            <far>1.1</far>
            <horizontal_fov>1.5</horizontal_fov>
            <aspect_ratio>1.2</aspect_ratio>
          </logical_camera>
```

Thank you

barrygxwan, 2021-03-14:
Agree, we also have this problem.

dan9thsense, 2021-03-14:
Thanks for considering my concerns and also for the interim method for continuing to use the original logical camera parameters.

Pavel-P, 2021-04-01:
The latest update addresses some of the camera coverage issues. We provided a new sample layout that takes advantage of the improved camera ranges to help ease the transition a bit. The main motivation for these changes was precisely the cost analysis (or lack thereof) for factoring-in the gantry camera. Previously, the base number of logical camera's needed to fully cover the bins/stations/AGVs was 16, leading to a cost factor of 8000. This cost factor is always going to be many times larger than using solely the gantry camera if it's priced at all reasonably. It's essentially impossible to make-up the cost factor difference through a higher efficiency factor.
The goal of the coverage changes is to lower the base camera count to something more reasonable (6) while still keeping the relative effectiveness of different strategies the same. A logical camera approach will still be more expensive than an RGB-D approach, which is more expensive than using solely the gantry camera. Limited camera approaches (using only over-bin camera's, for example) are also still viable, the savings are just reduced slightly. Essentially, if you were satisfied with your strategy before, these changes shouldn't force you to change to anything else.
I'm closing this issue for now since lowering the cost factor range is critical for keeping the core sensor options open.

denbar-robotics, 2021-04-01:
I think you have no idea how much work this change turned into wasted time.  It is extremely frustrating.  This determination should have been made before the competition was released.

Pavel-P, 2021-04-02:
After reviewing this, we realized there were certain sensor configurations we didn't account for in our own analysis that were disproportionately affected by the changes to the coverage. This wasn't the intention of the coverage changes and so we'll be rolling back these changes immediately. This one was on us, and apologies for the confusion and added stress.

We'll still be trying to address the issue of extremely disparate cost factors in a separate update, but we'll be looking into far less intrusive means of doing this.

closed 2021-04-02 by Pavel-P