usnistgov/ARIAC #79: proruct drop

YunpengHe, 2021-03-14:
When we using 'sample_assembly.yaml' and run the code 'rosservice call /ariac/agv1/to_as1' , the product of blue pump on agv1 will drop. I think there may be a problem with the new code. So we ask for your help.

zeidk, 2021-03-14:
Please do not use this service directly but use /ariac/agv1/submit_shipment "as1" "order_0_kitting_shipment_0" instead (see [here](https://github.com/usnistgov/ARIAC/blob/master/wiki/documentation/competition_interface_documentation.md#process-management))

closed 2021-03-14 by zeidk

barrygxwan, 2021-03-15:
if this order is a assembly shipment on 'as2', we need deliver agv to 'as2' first, and then assembly. in this situation, if we call the service  /ariac/agv1/submit_shipment "as2" "order_0_kitting_shipment_0", will the system think we submit a wrong shipment?