usnistgov/ARIAC #81: gantry arm controller errors

dan9thsense, 2021-03-18:
When the gantry arm has a part attached, it becomes very fragile to movement, generating timeouts.  For example, before attaching to a battery part, the wrist3 joint has no problem rotating.  Right after attaching, sitting in mid-air, the same joint experiences a failure that leads to a timeout, even when only moving a small amount.  I tried increasing the goal tolerances to no avail.  This is using the very simple "group.go" command, with no other constraints.  The move usually fully executes, so I thought it was an issue with goal tolerances but I tried increasing the goal tolerances to no avail. 

[ INFO] [1616030806.455647253, 142.203000000]: Planning attempt 1 of at most 5
[ INFO] [1616030806.456085324, 142.203000000]: Planner configuration 'gantry_arm' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1616030806.456244241, 142.203000000]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1616030806.470791437, 142.206000000]: RRTConnect: Created 5 states (2 start + 3 goal)
[ INFO] [1616030806.470842094, 142.206000000]: Solution found in 0.014675 seconds
[ INFO] [1616030806.495034671, 142.210000000]: SimpleSetup: Path simplification took 0.024138 seconds and changed from 4 to 2 states
[ERROR] [1616030814.425471644, 143.482000000]: Controller is taking too long to execute trajectory (the expected upper bound for the trajectory execution was 1.258976 seconds). Stopping trajectory.

zeidk, 2021-03-18:
Thanks, I am looking into this. Are you using moveit_commander?

dan9thsense, 2021-03-18:
using moveit commander

zeidk, 2021-03-19:
I think I managed to fix it in different places. I just need to try it with the other parts to make sure the part collisions are correct. I will then push an updated version.

dan9thsense, 2021-03-19:
I'm finding it also happens with the kitting arm when parts are oddly attached, such as when just a corner of the part is grasped.

dan9thsense, 2021-05-02:
Fixed.

closed 2021-05-02 by dan9thsense